tomcl/issie #14: Standard wire intersection notation?

MarcoSelvatici, 2020-07-10:
[image] **Issue by [ms8817](https://github.com/ms8817)**
_Friday Jun 19, 2020 at 14:28 GMT_
_Originally opened as https://github.com/MarcoSelvatici/DEflow/issues/47_

----

At the moment the applications uses:
- wires that are crossing but not touching: straight lines.
[image]
- branches: curved lines.
[image]

One may consider moving the standard notation with a dot instead.
Draw2d actually offers a connections router that has this capability (`CircuitConnectionRouter`) but it is not interactive (wires cannot be manually moved). One could write their own `InteractiveCircuitConnectionRouter`.

tomcl, 2022-02-12:
We lost curved lines with the new F# draw block. So we have now only one method out of 3!

tomcl, 2022-04-16:
fixed v3.0.0

closed 2022-04-16 by tomcl