tomcl/issie #17: Custom combinatorial components with Verilog.

MarcoSelvatici, 2020-07-10:
[image] **Issue by [ms8817](https://github.com/ms8817)**
_Friday Jun 19, 2020 at 22:07 GMT_
_Originally opened as https://github.com/MarcoSelvatici/DEflow/issues/55_

----

It would be nice to be able to define combinatorial components using a subset of Verilog.

Steps:
- [ ] Setup a text editor. The best way is to probably wrap the  `react-simple-code-editor` and display in a popup like the memory editor.
- [ ]  Write a parser for a Verilog combinatorial subset.
- [ ] Create a function to use the parsed Verilog as a reducer for a simulation graph node.
- [ ] Join everything together and test it.

tomcl, 2022-04-21:
Verilog subset implemented.

For ok intro to Verilog try https://www.allaboutcircuits.com/technical-articles/describing-combinational-circuits-in-verilog/

A competent combinational circuit subset of verilog should allow:

A single module containing module_items.
each module-item is one of
- assign statement `assign a = b & c` // here a must be declared as an out in the module header
- wire statement (which is assign + declare the lhs (a) as a wire) `wire a = b & c`
- lots of operators (see verilog reference) used on rhs of assignments
- signals of varying width (like Issie)
- slices (to select subsets of busses) used on rhs and lhs of assignments

in addition the following module items would allow much more flexible combinational descriptions (but this would be beyond the MVP):

- reg declaration statement (needed to for always blocks then do not directly drive reg outputs)
- always_comb statement
   - can contain if and case arbitrarily nested, and sequential statements (as in a programming language)
   - all sequential assignments must be blocking, and therefore to reg variables or outputs
   - all outputs (things on lhs of any blokcing assignment) must be assigned values on all paths through if and case

### Semantic checks

in addition to parsing verilog, there are a number of obvious semantic checks which should be made, with errors auto-corrected:

- Every identifier used must be declared (as output/input or internal)
- Every identifier used muts be driven exactly once - except module inputs which must not be driven at all
- Widths should always match across assignments

tomcl, 2022-04-22:
### Outputs

- The Verilog combinational blocks should have associated simulation-ready abstract syntax trees, which are evaluated by the fast simulation reducer
- Each block must have available the input and output widths of each port (these are needed by width inference)
- Each block must have available which inputs each output depends on. This info is used by the fast simulator. In fact, to make this clean, outputs with separate inputs should be treated by the simulator as multiple FastComponents which are reduced separately, since they can be placed at different points in the reduction order.
- Each block must generate its Verilog output. This should look quite similar to the block text! It should be inserted into the output file by Verilog.fs.

tomcl, 2022-04-23:
Combinational Verilog blocks should surface constant parameters. This will allow blocks to be written which fit different bus widths. Ideally this must integrate with #147 so that on a design sheet these parameters can be bound to sheet parameters. Since #147 assumes this will happen with other parametrised components it should similarly happen for the verilog block parameters which are surfaced. Probably these should be written as module parameters.

tomcl, 2022-05-10:
Planning

(1) get simple react-simple-code-editor to work with F#/Elmish interface. Put its state in model. Use message when state changes. 
(2) get parser generator (CLI) working. Find & translate (if needed) compatible online grammar
(3) work out subset of grammar needed for verilog input (PoC and expected endpoint)
(4) add JS parser to Issie, create interfaces to/from F# via JSON etc for the required verilog subset, check POC parser works
(5) generate AST from parse tree. use AST to do syntax highlighting on Issie in editor component
(6) Right a FastSim datatype-compatible evaluator for ASTs
(7) add Verilog component, connect to editor etc
(8) add fastsimulation component generator. One Verilog component will result in multiple FastSim components, one per output. Each component will run in FastSim reducer by calling the evaluator. Get Fast sim working with these
(9) add verilog output from the verilog component
(10) use parse output to do autocomplete and nice errors in edit component

apantelopoulos, 2022-09-01:
Closed with latest PR from 'verilog-component' branch

closed 2022-09-01 by apantelopoulos