tomcl/issie #23: Insert components via drag & drop.

MarcoSelvatici, 2020-07-10:
[image] **Issue by [ms8817](https://github.com/ms8817)**
_Wednesday Jun 24, 2020 at 15:49 GMT_
_Originally opened as https://github.com/MarcoSelvatici/DEflow/issues/64_

----

This should help: https://www.w3schools.com/tags/att_global_draggable.asp
Also, [Brainbox](https://github.com/freegroup/brainbox) uses the same drawing library and supports drag and drop, so peeking at how they do it may be helpful.

MarcoSelvatici, 2020-07-10:
[image] **Comment by [tomcl](https://github.com/tomcl)**
_Wednesday Jun 24, 2020 at 19:24 GMT_

----

As an interesting alternative, might consider insert components via intelligent heuristic placement!

tomcl, 2020-10-25:
With nice component placement now I'm not sure that drag and drop is any better than what we have! Closing

closed 2020-10-25 by tomcl