tomcl/issie #3: Add waveform visualiser in simulation.

MarcoSelvatici, 2020-07-10:
[image] **Issue by [ms8817](https://github.com/ms8817)**
_Friday Jun 19, 2020 at 13:15 GMT_
_Originally opened as https://github.com/MarcoSelvatici/DEflow/issues/32_

----

Simulation output should be visible via a waveform generation interface.

Stubs to do so have been created in commit: `eeaf2192a8e10e7ffbebec8f7ccf62db308d8d11`.
These stubs focus on unrolling a simulation for many clock ticks, without feeding combinatorial inputs (useful for example to simulate a CPU or a state machine).

All the functions except for feedNClockTicks can be reused to create a combinatorial logic waveform visualiser. In other words, all the inputs combinations and their corresponding outputs are shown.

MarcoSelvatici, 2020-07-10:
[image] **Comment by [tomcl](https://github.com/tomcl)**
_Friday Jun 19, 2020 at 22:39 GMT_

----

NB this is mentioned in project report Ch 9.1

A less satisfactory, but quicker for high functionality, solution would be to use an existing waveform viewer such as gtkwave, which is available in executable form for macos, linux, windows and can be driven via command line from a vcd file.

http://gtkwave.sourceforge.net/gtkwave.pdf

The work would be:
(1) output .vcd file with simulation output data. VCD files are simple and human readable, although not space efficient.
(2) invoke gtkwave on the file as an external process to run simulation.

this is clearly not as fast or smooth as built-in simulation - but it would be quite quick. One disadvantage is the gtkwave look and feel which we would have no control of, it may be too complex, it will not be compatible with the rest of the app.

tomcl, 2020-10-25:
Done by edoardo over the Summer!

closed 2020-10-25 by tomcl