tomcl/issie #6: Improve Simulator performance.

MarcoSelvatici, 2020-07-10:
[image] **Issue by [ms8817](https://github.com/ms8817)**
_Friday Jun 19, 2020 at 13:34 GMT_
_Originally opened as https://github.com/MarcoSelvatici/DEflow/issues/38_

----

Refer to the final report to read about the performance of the simulator, and how it could be improved (sections 8.3 and 8.3.1).
#37 can also help improving performance.

MarcoSelvatici, 2020-07-10:
[image] **Comment by [tomcl](https://github.com/tomcl)**
_Wednesday Jun 24, 2020 at 12:25 GMT_

----

To amplify on that, persistent memoisation could be used, orthogonally to other speedups, at every design sheet edge. That would mean the typical repeat simulations done when testing would run much faster, and it would scale to large designs.

To do this would require a sheet simulation interface that is memoisable, making explicit the internal D flipflop inputs that control state (but that is easy, and maybe exists as is).

Memoisation info could be stored with timestamp with old samples discarded based on space limits.

For small components composed of gates, e.g. a custom carry generate/propagate unit of  up to 8 bit size, complete memoisation of all functionaily is useful. In this multiple instances of the same block can share memoisation data to advantage.

tomcl, 2020-11-15:
The basic simulator can be improved by using mutable arrays for the (updatable) inputs, and taking reducer inputs direct from this.

The recursive simulation algorithm using lots of maps is pretty good. Maps tend (strangely) to be faster than lists on JS, though a bit slower under dotnet.

One miss in the current algorithm is the propagation that waits till all inputs have come before triggering and output. This is mostly irrelevant because in clocked designs all inputs are always present, but it adds complexity. In an array based simulation that would go.

You will get a speed up of maybe 20X (ignoring reducer time) using mutable arrays for the propagated input values and the components (with component ids and ports turned into array indexes) and arrays for the mappings from output to input ports. That is because under JS arrays mutated are about 20X - 100X faster than maps updated. That would make the code much less pleasant, but the same structure would work. 

Component Reducer function time can be improved by replacing the bool list WireData type by int list, using ints for up to 32 bits per int, and using bit operations (fast in js) to implement logic functions.

I can’t see much advantage in replacing the recursive structure by a flattened system, and it would be much more work, and less easy to use.

memoisation would be fun. You could look at the graph and find maximal subgraphs that have number of state + input bits < some limit, and number of output bits not too large - though in terms of cost the number of outputs does not much matter.

closed 2021-05-13 by tomcl