tomcl/issie #7: Explicit global clock component.

MarcoSelvatici, 2020-07-10:
[image] **Issue by [ms8817](https://github.com/ms8817)**
_Friday Jun 19, 2020 at 13:47 GMT_
_Originally opened as https://github.com/MarcoSelvatici/DEflow/issues/39_

----

It may be beneficial for first year learning outcomes to be able to explicitly handle the global clock (currently, it is assumed that every component is implicitly connected to it).

From the report (section 5.6.1):

> It is also arguable that ﬁrst year students should explicitly handle and connect clock signals, instead of having them implicitly connected by the application. This decision has no impact on the simulation logic, as it would not change the fact that each synchronous component is connected to the global clock. Hence, this is implementable by introducing an explicit ‘global clock generator’ component that can be added to the diagram and connected to the clock ports of the synchronous components. Such component and its connections can be then easily removed from the diagram state that gets passed to the simulation logic, which will assume each synchronous component is implicitly connected to the global clock. In the interest of time, I decided to focus on more essential features.

MarcoSelvatici, 2020-07-10:
[image] **Comment by [tomcl](https://github.com/tomcl)**
_Friday Jun 19, 2020 at 18:51 GMT_

----

You don't need a component. Just a requirement to connect all clock signals to an input. Then infer that input is a clock in simulation and use it. Best not to have a specific clock name for clock input since it will be got wrong sometimes leading to silly errors? Whereas check that clk only connects to other clk with error if anything else, and then make it clock, would do. Unconnected clk would then be an error. A switch would allow all clk to be disconnected and clocks inferred instead, as now

MarcoSelvatici, 2020-07-10:
[image] **Comment by [ms8817](https://github.com/ms8817)**
_Friday Jun 19, 2020 at 19:36 GMT_

----

Using a normal input component rather than an explicit 'clock generator' component is fine too.

tomcl, 2022-04-16:
We choose not to do this.

closed 2022-04-16 by tomcl