tomcl/issie #9: Show list of recently opened projects in 'Open project' menu.

MarcoSelvatici, 2020-07-10:
[image] **Issue by [ms8817](https://github.com/ms8817)**
_Friday Jun 19, 2020 at 13:56 GMT_
_Originally opened as https://github.com/MarcoSelvatici/DEflow/issues/41_

----

This requires a form of permanent storage.

One could simply save a file into a known location such as `~/.DEflow/settings`. Doing so should be possible with the functions in `FilesIO.fs` (or slight modifications of them).

Alternatively, Electron may provide some mechanism to do so in a standard way. Worth investigating.

MarcoSelvatici, 2020-07-10:
[image] **Comment by [tomcl](https://github.com/tomcl)**
_Friday Jun 19, 2020 at 18:47 GMT_

----

https://www.npmjs.com/package/electron-settings

Here is how it is used in Visual2. you need to be a bit careful parsing persistent settings to make sure that if they are unexpected (e.g. due to a different version of app having previously been installed) nothing bad happens. Hence all the checking below maybe over the top.

```
type VSettings = {
    EditorFontSize : string
    SimulatorMaxSteps : string
    EditorTheme : string
    EditorWordWrap : string
    EditorRenderWhitespace : string
    CurrentFilePath : string
    RegisteredKey : string
    OnlineFetchText : string
    }

/// A reference to the settings for the app
/// persistent using electron-settings
let settings : obj = electron.remote.require "electron-settings"

let mutable vSettings = {
    EditorFontSize = "16"
    SimulatorMaxSteps = "20000"
    EditorTheme = "solarised-dark"
    EditorWordWrap = "off"
    EditorRenderWhitespace = "none"
    CurrentFilePath = Fable.Import.Node.Exports.os.homedir()
    RegisteredKey = ""
    OnlineFetchText = ""
    }

let checkSettings (vs : VSettings) =
    let vso = vSettings
    try
        let checkNum (n : string) (min : int64) (max : int64) (def : string) =
            match int64 n with
            | x when x > max -> def
            | x when x < min -> def
            | x -> x.ToString()
        {
        vs with
            EditorTheme =
                match List.tryFind (fun (th, _) -> (th = vs.EditorTheme)) themes with
                | Some _ -> vs.EditorTheme
                | _ -> printfn "Setting theme to default"
                       vSettings.EditorTheme
            SimulatorMaxSteps =
                checkNum vs.SimulatorMaxSteps 0L System.Int64.MaxValue vso.SimulatorMaxSteps
            EditorFontSize =
                checkNum vs.EditorFontSize minFontSize maxFontSize vso.EditorFontSize
            CurrentFilePath = checkPath vs.CurrentFilePath
        }
    with
        | _ -> printf "Error parsing stored settings: %A" vs
               vs

let setJSONSettings() =
    let setSetting (name : string) (value : string) =
        printf "Saving JSON: %A" value
        settings?set (name, value) |> ignore
    printfn "Saving settings to this PC: %A" vSettings
    setSetting "JSON" (Fable.Import.JS.JSON.stringify vSettings)

let getJSONSettings() =
    let json = settings?get ("JSON", "undefined")
    printfn "Getting settings"
    match json = "undefined" with
    | true ->
            printfn "No JSON settings found on this PC"
            setJSONSettings()
            vSettings
    | false ->
        try
            let vs = (Fable.Import.JS.JSON.parse json) :?> VSettings
            vs
        with
        | e ->
            printfn "Parse failed: using default settings"
            vSettings
```

tomcl, 2022-04-19:
Now that we have `UserData` in Model persistent it would be very easy to add this.

tomcl, 2022-04-22:
closed by 3.0.0b.4

closed 2022-04-22 by tomcl