tomcl/issie #42: Problem with icon on Linux

MarcoSelvatici, 2020-08-27:
Building the app on Linux (with `yarn start`), I get the following error:

[image]

which looks very similar to https://github.com/electron/electron/issues/24221.
As suggested in that thread, this can be likely resolved by using a NativeImage:
https://github.com/fable-compiler/fable-import/blob/e3764b4823894dbbb510db029ca9ea346b78c5c7/Electron/Fable.Import.Electron.fs#L133

Commenting out:
https://github.com/tomcl/issie/blob/1c154ce3cfebfa96ee8979432691500e5c7f956c/src/Main/Main.fs#L72
allows me to work around the problem and run the application anyway.

tomcl, 2020-08-27:
The alpha release will not be run on linux, but I'll add a workaround to delete icon on linux if no time to do anything else - assume while this issue remains open the fix above it needed.

tomcl, 2020-08-27:
Preliminary fix so icon is not set on linux.

closed 2020-08-27 by tomcl