tomcl/issie #53: Memory editor should have higher z-index

MarcoSelvatici, 2020-09-07:
Currently, this is what you get when you open a memory editor.

I think it can be fixed by increasing the z-index of the memory editor view.

[image]

Question:
why has the z-index of the right hand column been increased to 30? @tomcl

tomcl, 2020-09-07:
> This email from notifications@github.com<mailto:notifications@github.com> originates from outside Imperial. Do not click on links and attachments unless you recognise the sender. If you trust the sender, add them to your safe senders list<https://spam.ic.ac.uk/SpamConsole/Senders.aspx> to disable email stamping for this address.

Yes, I think you are right. I struggled for a while making code that would move the middle divider properly – probably then. It should go down.

We have modal popups (fixed Fulma 40)
Rh side (can’t now remember but I think needed to be higher than LH side to avoid LH navbar showing when divider is moved)
Non-modal popups. ? I’ll try reducing RH side z.

I also have an issue about loading where I am not properly reloading (as loadedcomponent) sheets when the current file is changed. That was due to a slightly unpleasant bit of code managing auto-save and asking whether to select auto-save or saved version when files are reloaded.

I should fix both today.

I’ve been adding tooltips to the properties tab which is fine except then I need to ensure it is not recreated too often – because that means tooltips recreated. That means a lazyView with the right equality function.

Best wishes, Tom

[image]<https://user-images.githubusercontent.com/26204867/92388710-06929400-f118-11ea-93c4-7c5d1c0408bf.png>

tomcl, 2020-10-25:
OK, I've redone the z-indexes and all is now good I think - closing this.

closed 2020-10-25 by tomcl