tomcl/issie #60: Rom editor should show leading zeros

pclemow, 2020-10-25:
ROM editor should retain leading zeros for clarity.

Eg 0x04 rather than 0x4 when data width is 8 bits

tomcl, 2020-10-25:
Agreed, some care needed since rom data can be any width but it is still easy, i'll do it in next week.

tomcl, 2020-10-29:
Turned out to be high time due to problems in the original design other than leading zeros. now fixed, with a more MVUish design, and the ability to view and edit a 16 location area of memory anywhere in address space,

In addition the implementation of memory has been improved. Issie can now correctly simulate very large RAMs (e.g. 40 address bits) as long as the number of non-zero locations is fairly small.

closed 2020-11-08 by tomcl