tomcl/issie #62: If the loaded sheet contains an error then the "New Project/Open Project" dialogue doesnt close

pclemow, 2020-11-08:
If, when starting issie, a sheet is opened and instantly creates an error (for example, each input needs exactly one driver) then it wont open and the file and wont close the dialogue. This means you can softlock projects.

I have attached a project I seem to have softlocked this way. I was trying to debug the "block names match" error at the time so it might be trying to load straight into waveform mode.

[Phi0.zip](https://github.com/tomcl/issie/files/5507108/Phi0.zip)

tomcl, 2020-11-08:
Ok. I'm a bit surprised, this should not happen. I have improved the state transitions a bit in v1.0.4 so this is probably fixed.

Nope. Looking with dev tools this is a Draw2D infinite loop leading to a stack error. I have already put intrlocks to recover from Draw2d errors on connection loading. This is the first time I've found one on connection loading. i'll fix it. However, the corrupt (= causes draw2d bug) component will be deleted. there will be an error message indicating that, so not too bad.

Draw2d is a great library - but not 100% stable. long-term we will create our own draw library re-implementing the bits of D2D that we use. That will however be a big job.

Expect fix shortly - assuming what I'm expecting to work will work.

pclemow, 2020-11-08:
Cool. Looking at the DECA team, it looks like a student has this as well (though I guess he'll be on 1.0.3, rather than the 1.0.4 I was playing with)

tomcl, 2020-11-08:
OK, the latest version of 1.0.4 (currently Windows - but macos will be updated in 5 min time) provides extra protection against draw2d bugs on load. it seems to solve your problem - the project loads with one lost connection. Have a look and let me know.

tomcl, 2020-11-08:
OK - in addition: I've removed the check on split connections without names - since these do not nicely highlight (the highlighted area is inside the component boundary and they do not have any such!).

I found pretty quickly which was the bad connection - it crashed issie whenever i tried to move it. I deleted everything around it and remade. 60s. All is good now. Mended project enclosed.
[Phi0-mended.zip](https://github.com/tomcl/issie/files/5507324/Phi0-mended.zip)

closed 2021-05-13 by tomcl