tomcl/issie #63: Error "2 Items have the same name" is confusing and tricky to debug.

pclemow, 2020-11-08:
The attached project was giving me a "matching names" style error and said to change the highlighted component. There was no highlighted component so I do not know where to look.

Unfortunately this may also be softlocked by the other issue I have raised so the project may not be that useful, sorry!

[Phi0.zip](https://github.com/tomcl/issie/files/5507128/Phi0.zip)

tomcl, 2020-11-08:
:)

v1.0.4 highlights this

pclemow, 2020-11-08:
Im using 1.0.4 ...

I suspect the error was in a subsheet but no idea which one as I got locked out of the project.

tomcl, 2020-11-08:
as of 11:15 todat both platforms v1.0.4 should deal with this. the problem is that splitwire was previously checked for having non-null and distinct name - but splitwire components do not highlight, having no interior. This should be mended but it would be difficult (lots of JS and draw2d checking to find out how).

So I've disabled this check and now all is good.

closed 2021-05-13 by tomcl