tomcl/issie #66: Start wavesim with initial input values

Jpnock, 2020-11-30:
### What?

Add a "start wavesim with these initial values" button on the step simulation window, in order to enable the user to start the waveform simulation with user-specified initial values for inputs rather than zeros.

### When?

It should only appear on step 0 of a clocked design.

### Why?

A few times I've found myself swapping inputs for constants to replicate this functionality, then having to revert the change afterward.

Certainly not high priority at all but potentially a nice feature to eventually have.

Special thanks to @tomcl for your input on this feature request.

tomcl, 2022-04-19:
I've changed tags on this because although doing this badly would be easy, doing it with a proper UI is not so straightforward. In addition, the top-level simulation UI is a bit of a mess at the moment so this should get folded into wider simulation overhaul.

tomcl, 2022-05-04:
#160 would partially solve this issue.

PR #181 (2022-06-30): Add support for Input components to be simulated with non-zero values
Closes #66 and #160.
Closes [jzzheng22/issie/#17](https://github.com/jzzheng22/issie/issues/17)

This PR adds support for Input components to have a default value when they are not driven.

Default values can be set on component creation or in the Properties tab.

This requires the addition of the `ComponentType` `Input1`, with `ComponentType.Input` becoming a legacy component type.
Any old Issie sheets should automatically have their `Input` types converted to `Input` types on loading the saved file.
closed 2022-06-30 by jzzheng22