tomcl/issie #70: Inconsistency in bus select name checking

harry1453, 2020-12-13:
Here: https://github.com/tomcl/issie/blob/0858e10a287ed8d64cdbec32b15270d236fc58f8/src/Renderer/Simulator/CanvasStateAnalyser.fs#L428

BusSelect components are filtered out of the empty name check. But here:

https://github.com/tomcl/issie/blob/0858e10a287ed8d64cdbec32b15270d236fc58f8/src/Renderer/Simulator/CanvasStateAnalyser.fs#L441

They are not filtered out of the duplicate name check. This means that bus select components with empty names get the duplicate name error.

When there is only one bus select in the sheet, leaving it without a name works fine. Which leads me to believe that empty names are valid for bus selects.

If this is the case, I think they need to be added to the filter for the duplicate name check.

If this is not the case, I think they need to be removed from the filter for the empty name check. (The error message "An empty name is not allowed except for split and join" leads me to think this might be the case)

tomcl, 2020-12-13:
It is not entirely clear, the issue is whether duplicate names on bus selects lead to ambiguous or overly complex labels of waveforms. Still TBD. But I agree empty would be more usable.

harry1453, 2020-12-13:
Fixed in 771adfd14485c36fc6fda64c52f888d876a7bdbf

closed 2020-12-13 by harry1453