tomcl/issie #75: Possible issie QoL upgrades

Thomas73951, 2021-04-05:
# Issie QOL upgrades

Below are a few things, mostly quality of life upgrades, that would make issie better. With this I'm not saying by any means that issie is bad, simply that these are some things that I've noticed that would make using issie easier. 

I am aware that some things in this list are much easier to implement and/or are higher priority than others; I think it's still worth me making you aware of all of them. 

I am happy to elaborate on any points below if needed. 

### UI

- enter for green boxes (like add)
	- when doing things like adding inputs, I instinctively press enter to add that item, but nothing happens. Instead I have to go back to mouse and click on `add` 
- Allow constants to be loaded with bin and hex. 
	- currently the only way to put a value in a const is with dec. 
	- (Just displaying hex is fine)
- copy and pasting sections isn't particularly useful
	- as most of the component names have to be changed manually, which is quite fiddly. 
- when a wire label is selected, it would be nice to also highlight all the labels it links to (all others with the same name)
	- (and not in green as that's the colour used when selecting an item, it needs to be clear which one you clicked on)
	- this is mainly because on large sheets with lots of wire labels it can be hard to see what's connected to what. 
- add text to sheet for writing notes and stuff (like you can in ltspice)
- search on sheet. 
	- Searches all text on the current sheet and highlights the matches. an alternative to ^ 
- undoing a copy paste (with many items) removes items one at a time instead of all at once.
- keyboard shortcuts
	- these need changing to something more standard (`ctrl` + `z` for undo not zoom)
	- and need to reliably work.
- Copy paste ram data 
	- the ability, from within the ram editor window, to copy or export the contents of that ram 
		- the ability to export/import as a text file (or similar) would be great. It would solve the problem above, but also allow for easier entry of instructions. 
	- and then paste or import that into a different ram block on the same project or share with others. 

### Simulation/waveform

- ability to reorder elements in simulation tab
	- this exists in waveform but not simulation. 
- for it to remember the order in waveforms between each test
	- currently its a faff to reorder the list every time after changing something. 
- waveform - have all in bold preselected. 

### Performance 

(I believe all of these are already being worked on)

- switching between catalogue etc can take up to 10 secs. 
- switching between large sheets takes too long 
- waveform simulation sometimes takes 30-60 seconds (and not on a slow computer)

tomcl, 2021-04-10:
Thanks for this. Most of what is summarised here is expected to be complete by the end of June in a major upgrade. The UI issues (enter not add etc) will be folded into a UI review and get done if possible.

Just some points for now:

* As of v1.3.0 simulation is about 10X faster - and more importantly it now works!
* Current master (not yet versioned) has tab change faster. I *think* it solves the bug, but am not 100% certain. You can easily generate your own latest version binaries if you want to try that.
* Weird key shortcuts - this is an electron and maybe draw2d issue (it intercepts those keys) and not easy to change. It is thankless work trying to find out how to do this, and we have tried a bit, so no promises this will be sorted - unless someone wants to tackle this with a PR. If the problem is actually just draw2d then it will be fixed with the new draw block.
* Wire label net highlighting - agree this would be easy and is highly useful
*  Ram copy-paste etc. Yes this will be done somehow - Maybe by init file in project.
* Properly remember sim waveforms order - yes this got left out of previous dev and is not difficult.

And my pet hates:
* Load/change sheet takes too long. Should be nearly instantaneous
* Nice wires do not work (they crash issie)

Those are both fixed with integration of a new draw block, which will happen as soon as we can do it - but it is  a big job.

JEMerrick, 2021-05-04:
Hi, we have just released ISSIE v1.9, and I thought I would let you know about the current progress and the status of the suggestions you made in your feedback about ISSIE, as well as the future improvements we have planned.

# State of the suggestions:
- Enter for green boxes
  - In future todo list
- Allow constants to be loaded with bin and hex
  - In future todo list
- Copy pasting sections not useful
  - Done, see below
- Highlight all labels connected to selected wire label
  - In future todo list
- Add text for writing notes
  - Not planning on implementing currently
- Search on sheet
  - In future todo list
- Undoing a copy paste (with many items) removes items one at a time instead of all at once
  - Done, see below
- Keyboard shortcuts
  - Done, see below
- Copy ram data
  - Done, see below
- Reorder elements in simulation tab
  - In future todo list
- Remember order of waveforms 
  - Done, see below
- Preselection of waveforms
  - Done, see below
- Performance (all)
  - Done, see below

# Full list of changes

- New component
  - Viewer component can be added to any net on any sheet, and will allow both Step and Wave simulation to see the net it is attached to, even if this is a subsheet.
- Gates
  - New-style IEEE rectangular 2 input gates replace old-style different shape gates
  - See also future works
- Keybinds
  - Copy: Ctrl+C 
  - Paste: Ctrl+V
  - Undo: Ctrl+Z
  - Redo: Ctrl+Y
  - Save: Ctrl+S
  - Zoom in application: Ctrl+Shift+=
  - Zoom out application: Ctrl+Shift+-
  - Zoom in canvas: Shift+=
  - Zoom out canvas: Shift+-
  - Zoom in/out canvas: Ctrl+Mousewheel
  - Zoom canvas to fit screen: Ctrl+W
  - Select all: Ctrl+A
  - Cancel operation: Esc
- UI
  - Drag mouse left click creates a selection box
  - Holding Ctrl toggles selection
- Features / Bugfixes
  - Verilog output (experimental)
    - Circuit can be output as a single Verilog file for synthesis or Verilog 3rd party simulation
    - Current output file, pasted into online Icarus v10 Verilog simulator, prints out the simulation, seems to work on some big designs.
  - Simulation
    - New *fast simulation* released informally in v1.2.0
    - Fixes all previous bugs (completely)
    - Robust algorithm - designed for quick fixes if there are any alpha bugs
    - Approx 10X faster than old simulation.
  - Manual wire routing
    - Left click and drag a wire to move it manually
    - Move an attached component to reset to automatic routing
    - UPDATES TO COME: This current system is not ideal, and a better version of it is high on the priority list
  - Adding/pasting components
    - Adding a component will no longer spawn it at a random canvas position
    - Click to drop implemented
    - Upon selecting the component in the catalogue, move the mouse and click to place the component
    - When pasting, upon pressing Ctrl+V the copied components will be shown at the mouse location, and can be placed anywhere that does not overlap components
    - Overlapping components will be shown in red
  - Automatic label naming
    - Pasting copied symbols will automatically update the names by altering (or adding) the number suffix
  - Saving
    - Autosave has been removed as ISSIE no longer crashes randomly
    - The 'Save' button will be opaque and green if there are unsaved changes
    - Switching sheet will save changes automatically
    - If you try to exit the program with unsaved changes, a prompt will come up asking if you wish to close without saving, or go back and save
  - Sheet corruption fixed
  - Undo/Redo
    - Now undoes/redoes an entire pasted section rather than 1 by 1
  - Memories ROM/RAM
    - ROM or initial RAM contents may be exported to files, or imported from them
    - New signed and unsigned multiplier ROM option
  - Waveform simulator
    - Remembers previous order
    - Preselection
- Lag/latency
  - Vastly improved most slow operations (right hand tab, load project, change sheet, waveform simulation etc)
  - Throttling has been added so that in the event of slow hardware or if you are doing large operations e.g. moving many components at once, the program will drop frames in order to prevent lag. In this case it will appear more stuttery, but be more responsive and useable. Most people should not notice any stuttering problems.

# Future works

My current list of features which will be relevant to users, no guarantees that all of these will be done, and in no particular order:

- Wire routing algorithm improvements
- Option to switch between new IEEE and old logic gate symbols 
- Option to switch between wires with jumps and wires with solid circles
- Keyboard shortcut for common components (e.g. WireLabel)
- Symbol manipulation (scale/rotate symbols)
- General UI improvements
  - Allow enter to confirm popups
  - Allow underscores in names
- Add back button to simulator
- Improving time for loading projects / sheets
- Colour options / Accessibility
- Automatic updating custom components
- Allow placing multiple components
  - e.g. Keep placing AND gates on mouse click until Esc key is pressed
- Allow constants to be loaded with bin and hex
- Wire label highlighting all connected labels
- Reorder elements in simulation tab
- Search for name of component on sheet

If you have any opinions about this, or ideas that are not here, please let me know by creating an Issue on GitHub or (Imperial students) posting on Piazza.

tomcl, 2021-05-05:
As of v1.9.4 a major overhaul of constant components now satisfies the requirement above and more.

vaclavpavlicek, 2021-05-06:
Hello! Today morning, I tried using version 1.9.4 of Issie in Wine on Ubuntu that was running in the Virtual Box. After placing a component into a sheet I was not able to drag and move it somewhere else. I am adding a screencast of the bug in the attachment. Could you please fix it? In case you need more information about the bug please let me know.

https://user-images.githubusercontent.com/6002134/117260696-a0e5db80-ae4f-11eb-8708-5089dcddc06c.mp4

JEMerrick, 2021-05-06:
Hi, thanks for letting me know. In future it might be better to make a new issue on GitHub for separate bugs to keep things tidy. I think I have fixed what caused your issue, I had a look and it seemed to be caused by attempting to view the waveform tab when the circuit has errors. Did you by any chance try to press the waveform button and it returned a circuit error before this happened (just so I can confirm the bug I fixed was indeed what caused the situation you reported)? If so this will be fixed in v1.9.5 which will shortly be uploaded.

tomcl, 2021-05-06:
Bugs reported by EEE students can be handled outside of Github issue on Teams or by e-mail, or as conversation on a github issue here. the main thing is the conversation. We will do our best to understand and answer bugs - but this often requires a Q&A so we can understand precisely what is the issue.. Github does this quite well, but as Jo has said, it is helpful to separate bugs so each issue stays on topic. This issue is a general - "what features can be added?" topic.

JeffreyWong20, 2021-05-08:
Is there any possible ways which allows me to change the size of the block, cause it seems to me a lots of character(name) couldn't be displaced properly due to the size.[image]

tomcl, 2021-05-08:
This is due to be fixed pretty soon - talk to Jo about when.  We are aware!

tomcl, 2021-05-09:
v1.9.7 ticks the "make it easier to change custom components" wish list item.
Currently it will automatically update the instances, keeping instance connections when width is changed, or for connections not altered by a change

To follow later (possibly) "guessed" mapping of new to old ports with different name and same width (e.g. when you rename a port). That will allow renaming of ports without any manual reconnecting.

JEMerrick, 2021-05-12:
> Is there any possible ways which allows me to change the size of the block, cause it seems to me a lots of character(name) couldn't be displaced properly due to the size.[image]

Yes, this has been fixed and will be available in the new update today. I have made it so that when you add the custom component it will be scaled accordingly to fit all labels/names. Thank you for bringing it to my attention :)

closed 2021-10-31 by tomcl