tomcl/issie #99: Issie Polishing

tomcl, 2021-05-12:
This issue documents publicly the small Issie look and feel features that could fairly easily be changed or added. It does not cover major drawing functionality due to be upgraded, but might e.g. cover the coloring of wires and busses.

Comments below will get folded into this list, and/or possibly help to establish priorites:

- Zoom-to-fit (missing and easy to do)
- Right-mouse-click or space-mouse-click shortcuts to object specific actions e.g. open properties on object, go to sheet (if custom component) etc.
- Colors to easily examine wire label connected nets
- More components (N bit counter, bigger multiplexers) - currently low priority
- Color on diagram - busses, as in Issie. Inputs, outputs, wire labels? the trade-off here is that color can be informative but also can be distracting
- Search, of different kinds.
- Different sized symbols. Overall symbols are not yet as nice as they were in Issie.

aSamHF, 2021-05-18:
I was told that I should submit my comments here, so here goes:

1) The merge wire block has no indication of whether the top of the bottom input is the most significant bit. Previously there was a visual indicator of where bits went when using merge and split wires which were very helpful for checking mistakes. If this feature does have to go, updating the component description would be nice for convenience.
2) When adding in sub-sheets, if the default name of the block could include numbers, it would just be convenient (e.g. shift1 or shift2 rather than just shift), this is very minor though.
3) When entering a name for a component e.g. a wire label, adding it to the sheet causes lowercase letters to be automatically capitalised in its name (which I personally quite like), but when editing the name in properties, this automatic capitalisation disappears.
4) When copy-pasting components (very glad it's a thing bound to ctrl c and v), components are given new names in all cases. While I see why this could be desirable for preventing errors, it is particularly annoying when it comes to wire labels as they need to be the exact same name in order to properly function. If an exception to the new name generation could be made for wire labels I'd be very happy.
5) Might just be me, but I personally liked the select line of multiplexers being at the bottom. Was fairly convenient for formatting.

Thanks very much for the ongoing support.

tomcl, 2021-05-18:
v2.0.8 keeps wire labels the same. Is this better? It will fit some use cases, be wrong for others (see release notes).

tomcl, 2021-10-31:
* 1,3,4 fixed
* 2 - remains - i'll keep this issue open till it gets done.
* 5 - due for fix Spring 2022. It is very complex to do this because it requires the auto and manual routing all to work for different types of wires.

tomcl, 2022-04-16:
done v3.0.0

closed 2022-04-16 by tomcl