tomcl/issie #108: Sheet focus bug

bet20ICL, 2021-07-20:
Occasionally Issie will 'push' the focus of the sheet to the top left corner. It is pretty jarring when it happens, especially when working on a large sheet, it is time consuming to navigate back to the place you were working before.

tomcl, 2021-07-25:
Yes, I've noticed this. It is not so easy to replicate - if you find a way post this here and I can investigate it directly.

Otherwise I think it will be covered by #107 where I suspect the auto-scroll logic sometimes goes wrong at high zooms. It needs a proper look at the zoom / scroll logic. This is not much code. but i'm not sure anyone understands it exactly at the moment because it has some nasty svg/html/react dependence.

This bug will definitely get fixed some time over the Summer.

tomcl, 2021-07-31:
I think it is now fixed in v2.1.1

Let me know if not.

closed 2021-07-31 by tomcl