tomcl/issie #110: Verilog output should have predictable and static port names

edstott, 2021-10-05:
Issie creates node names in the Verilog output by adding a machine-generated suffix to the user's port names.

The machine-generated names are also used as top-level port names in the Verilog output. This is inconvenient when instantiating the design in external Verilog or compiling it for an FPGA because the names in the instantiation or pin mapping file must be changed whenever the design is changed and exported again from Issie.

Ideally, the port names in exported Verilog should be identical to the names given by the user in Issie. However, the user's names may need to be cleaned or rejected if they are not compatible with Verilog syntax.

tomcl, 2021-10-06:
Good point. This is a bit messy to do precisely because of the cleaning issue and the fact that any change can introduce unintended clashes worst case. I'll have a look. It is definitely possible to clean so that suffixes are only generated very occasionally, when there is an unexpected clash. Also, I could attempt to preserve (no suffix) the I/O names, adding disambiguation only to internal node names except in the unlikely case that two extrenal port names clash. is it juts the external port names where this is an issue?

tomcl, 2021-10-06:
> This email from ***@***.******@***.***> originates from outside Imperial. Do not click on links and attachments unless you recognise the sender. If you trust the sender, add them to your safe senders list<https://spam.ic.ac.uk/SpamConsole/Senders.aspx> to disable email stamping for this address.
>
> Yes, just external ports for now. There may be similar problems with internal nodes in future if someone implements hardware debugging.
> One simple solution could be to create a top-level wrapper module in the Verilog output that instantiates the main module and maps internal names to user names with a static prefix to fix (some) illegal names.

Do we have a precise spec of what is a legal Verilog name? It could be narrower than any FPGA toolset, but must not be broader than that.

edstott, 2021-10-06:
[_a-zA-Z][$_a-zA-Z0-9]{0,1024}

$ here is a literal character, not a special character

Also need to exclude [keywords](https://www.intel.com/content/www/us/en/programmable/quartushelp/13.0/mergedProjects/hdl/vlog/vlog_file_reserved_words.htm)