tomcl/issie #115: There's no grid snapping when moving multiple components

shuyuan-liu, 2021-11-15:
Tested on ISSIE v2.3.1. When more than one component is selected and being dragged by the mouse, no grid snapping occurs at all. I think this might have been intended, since if not everything in the selection is on grid points to begin with it would be messy trying to determine which component to use as the reference for snapping the whole selection. Could anyone confirm my guess?

tomcl, 2021-11-16:
Yes. Frankly, the snap-to-grid is not ideal anyway. Better would be snap-to-component or snap-to-no-kinks-in-wire. those are more complex but would be more useful I think. I expect that done in the Spring HLP-student round of dev.

I will leave this open till then as a marker that snap-to-X could be improved.

tomcl, 2022-04-16:
Left open - the current snapping is very good. It will not work well with multiple components except when they are already aligned. There is a use case for this.

I'll close this when that additional small feature is added.

shuyuan-liu, 2022-04-16:
> the current snapping is very good

I think how good the current snapping is is down to personal preference and habit. I find it cumbersome having to manually move something near a grid line for it to snap, and would much prefer components to only ever be on grid points. But this is going off topic for this issue.

tomcl, 2022-04-16:
Have you tried the new snapping? I find it very ergonomic - it does not use gridlines, and snaps much better.

Snap to gridlines does not work: it puts too many constraints on component sizes, and anyway often does not snap where you want, to make connections. Which is why current snaps are all component-component or segment-segment.

Neatness helps make it easier to read schematics.

To deal with the manual alignment issue the next release will have align and distribute commands, on selected components, with some intelligence so it aligns the right things when you select a bit of extra stuff. You can snap the components at either end of a line and align and distribute for two aligned rows, for example.

Anyway, to address snapping with multiple components, when components are all aligned, I will enable snapping with multiple components selected along the access of alignment.

shuyuan-liu, 2022-04-17:
> Have you tried the new snapping?

Sorry I hadn't. Tried it and yes, being able to snap to output / input ports in very nice. Since components themslves don't snap to gridpoints anymore - I think this issue may be closed?

tomcl, 2022-04-17:
Closed by https://github.com/tomcl/issie/tree/v3.0.0beta.1

This adds, in addition to snapping, component align and distribute (as in powerpoint but with direction chosen automatically). To make convenient only one component type is aligned or distributed (whichever type has most symbols in the selection).

Also label rotation is added to shortcut tedious manual movement of I/O labels to left of right.

closed 2022-04-17 by tomcl