tomcl/issie #116: Cosmetic: App's icon has an opaque white background

shuyuan-liu, 2021-11-15:
issie.exe's icon has an opaque white background whereas most other file and folder icons in Windows do not. This makes the icon look slightly out of place when highlighted in Explorer as well as in the Start menu. Since [v2.3.0](https://github.com/tomcl/issie/releases/tag/v2.3.0) this also affects the icon in the taskbar.

[image]

[image]

[image]

tomcl, 2021-11-16:
:)

I don't have a clue about this windows icon stuff. I guess I need a png file with a transparent background. I will leave this open till somone (maybe me) feels inspired to deal with it. The png is: `/static/icon-1.png`.

tomcl, 2021-11-16:
I've turned this into an png with a transparent background. It still does not work. I may be missing something, e.g. the tray icon comes from somewhere weird in the build I'm not seeing

I'll keep this open but it is tricky to address and very low priority.

tomcl, 2021-11-18:
It is totally mysterious when this background is transparent and when not. I think I have fixed it v2.3.3 - but maybe not in shortcuts to old executables etc.

closed 2021-11-18 by tomcl