tomcl/issie #117: Improve error message on trying to create a project with a non-alphanumeric name

shuyuan-liu, 2021-11-15:
On ISSIE v2.3.2, if the user tries to create a project with a non-alphanumeric name, nothing happens except for the error message "Could not create a folder for the project." This can be baffling for a first-time user as filenames and folder names with spaces and underscores etc. are commonly supported in many other programs.

On this note, could someone explain why ISSIE limits project / sheet / component names to alphanumeric characters? I understand that special characters like `*` `?` etc could cause problems with parsing but can't think of why underscores and spaces aren't allowed. I've tried manually renaming a sheet file to have spaces in the filename; it simulated just fine and could be used as a component from other sheets too.

tomcl, 2021-11-16:
Better error message will be done in next release.

I agree the naming restriction could be relaxed a bit but there are issues about parsing in different parts of the code, also turning Issie names into Verilog names for Verilog output. In addition, component names conventionally do not have underscores, and would be problematic with spaces, because they would break up on schematics into two parts.

Perhaps allowing non-initial `_` characters would be useful? I will leave this open till that is done.

There is no obvious restriction on project naming needed, so this could easily be relaxed.

I'm not sure these relaxations - other than perhaps allowing `_` are very important?

shuyuan-liu, 2021-11-16:
Thanks for the reply @tomcl. Yes indeed I think _ is the most important one; other characters are nice to have but aren't necessary.

tomcl, 2021-11-16:
in v2.3.3 project names will allow _ and have proper error messages (that was a bug, they were meant to have proper errors anyway).

Sheet names allow ' ' already. I will add (v2.3.3) '_', and made the error message consistent.

Component names allow '_' already. There is no error because extraneous characters get removed. This is not quite as good as it should be but in practice I believe causes no problems.

tomcl, 2021-11-18:
Small changes to relax rules and improve error messages v2.3.3

closed 2021-11-18 by tomcl