tomcl/issie #125: Directly using ISSIE on network location (H:)

ChangLiu2022, 2022-01-26:
ISSIE 2.4.0 isn't functioning running directly from the H: drive using college computer. It behaves as follows and quits a while later.
[image]

tomcl, 2022-01-27:
Yes, that is correct. It is to do with networked PC  (I think).

**Workaround**

open command window, navigate to the top-level issie binary directory that contains issie.exe

run:

issie.exe --no-sandbox

ChangLiu2022, 2022-01-28:
> Yes, that is correct. It is to do with networked file system on PC (I think).
> 
> **Workaround**
> 
> open command window, navigate to the top-level issie binary directory that contains issie.exe
> 
> run:
> 
> ```
> issie.exe --no-sandbox
> ```
> 
> let me know if that works - it does work on a networked path on my PC.

It works! Thank you for sharing this. 

Find it easier to set up a shortcut with the ' --no-sandbox' at the end of 'Target:' line in its properties, so I don't need to type command every time I use issie. :)

closed 2022-01-28 by ChangLiu2022