tomcl/issie #126: Using Ctrl-C, Ctrl-V in ROM contents window also copies and pastes diagram components

shuyuan-liu, 2022-01-28:
Issie version: v2.4.0

Reproduced on OS:
- Windows 11 21H2 build 22000.438 
- Windows 11 21H2 build 22000.466
- Windows 10 21H2

Steps to reproduce:
- Place a ROM in a diagram;
- Select the ROM and in the Properties tab click on 'View/Edit memory content';
- Press Ctrl-C then Ctrl-V;
- Close the memory contents window;
- Now there is a "new" ROM waiting to be pasted under the mouse cursor.

I believe this is a significant issue because it's not uncommon to need to copy & paste instructions in a ROM, and although the result is harmless it's nevertheless annoying.

tomcl, 2022-01-30:
I will add this to the "good to do this Summer" issues. Or maybe for Spring groups going beyond their remit.

in fact as I understand it, this is just that diagram copy and paste is happening when not interacting with diagram - that can be fixed by filtering the key commands to sheet so they don't happen. It is slightly complex because of the way that electron intercepts cut and paste so Sheet gets these some weird way, which is probably why this happens. In addition Sheet model is isolated from the UI model - so info cannot be picked up directly. This can be changed (and should be) but the type refactoring is major.

shuyuan-liu, 2022-02-04:
Thanks @tomcl for the reply. I'll add that typing `_` (shift + `-`) and `+` (shift + `=`) in a pop-up window will cause the underlying diagram to zoom, probably caused by the same issue.

tomcl, 2022-02-04:
Closed in v2.4.3

tomcl, 2022-02-04:
fix broke something else.

tomcl, 2022-02-06:
Closed (for real) v2.4.3 tag 2.4.3.1

closed 2022-02-06 by tomcl