tomcl/issie #140: Problems exiting Waveform Simulator

tomcl, 2022-02-27:
Issie as of 2.4.4 has some problems when you start the waveform simulator but do not terminate it. This can happen (sometimes) by changing sheet or changing tab, and is particularly problematic when the simulation throws an error and you navigate away from this without closing it.

The symptoms are a white RH panel which cannot easily be changed, and/or sometimes inability to edit the schematic (you can't do this while simulating!).

Workaround: 
(1) navigate sheets to the sheet you simulated originally (usually the top-level sheet).
(2) Click the "waveforms" button to get back to the simulation, then close it properly. Both errors and waveform simulations need to be explicitly closed. The simulation must be closed in two steps, first to the edit panel, then close.

This UI is  regrettable and due reworking - it could be quite easily made robust so that it is impossible to navigate away from something that needs closing without closing it.

This will be high priority for developers: but we have no time scheduled for this immediately, so it may have to wait a few weeks. The above workaround always fixes problems.

tomcl, 2022-04-19:
There is no class use of Issie till next year - so this will be folded into sorting out the top-level simulation UI - either fixed as part of the Summer Term dev or in a post Summer dev sprint.

jzzheng22, 2022-06-20:
This should, I hope, be closed by #174.