tomcl/issie #141: Previous text in clipboard gets appended to name of wire label being copied

shuyuan-liu, 2022-03-04:
When copying and then pasting a wire label, any text already in the clipboard is pasted into its Component Name box if the box is highlighted (default), changing its name.

Steps to reproduce:
- Place a wire label
- Switch to the Properties tab
- Copy some text from either inside or outside issie
- Copy the wire label with Ctrl-C
- Make sure the Component Name box is highlighted, by clicking in it (should already be)
- Paste the wire label by pressing Ctrl-V

The name of the first wire label is changed.

Workaround: after copying a wire label click on empty space in the diagram to remove focus from the Component Name textbox.

Tested on issie 2.4.4.

tomcl, 2022-03-08:
Low priority, but worth fixing.

- Will try to do this before start of Summer Term
- We have a dev (Placement) after end of Spring Term who will maybe do that.

tomcl, 2022-04-16:
This could probably be fixed by removing focus from dialog boxes on ctrl-v. To be looked at later.

tomcl, 2022-04-20:
closed by fix in 3.0.0b3

There may still be some issues with other properties boxes, but the "default" wrong behaviour is now fixed.

closed 2022-04-20 by tomcl