tomcl/issie #143: Components should grow taller with new ports added

shuyuan-liu, 2022-03-06:
Currently components don't change size when new ports are added. If a component had many ports added, its symbol becomes very crowded and difficult to wire, as shown below.

[image]
[image]

I couldn't find a way to resize components in the GUI, and although one could edit the W and H values in a .dgm file, it's not a proper solution. This isn't a critical problem but still an annoying one (for me) in the EEP experiments where the DECODE block had many ports added.

tomcl, 2022-03-08:
Agreed. It is a consequence of an added feature (automatically changing symbols when you add ports to a sheet).

It will get fixed before Summer Term but alas the devs (that is me at the moment) have no time to spare till then.

Workaround:

- take a screenshot to make it easy to reconnect
- delete
- add
- reconnect

tomcl, 2022-04-05:
This will closed in v2.5.0 - see v2.5.0a1 for preview.

Although the auto-resize is now good, there is an argument for allowing manual resize as well from properties, since sometimes schematics are easier to read with some custom components larger than the minimum size.

We will keep this open till the v2.5.0 alpha dev is finished.

tomcl, 2022-04-16:
fixed v3.0.0.beta

closed 2022-04-16 by tomcl