tomcl/issie #144: Components are renamed in the properties tab when their name is numeric

1rre, 2022-03-08:
When a nameable component (such as a wire label, input or output) is given a purely numeric name (ie `4`) upon adding the component, the circuit diagram shows their name as entered, however in the properties tab, all these components are given the name `L` in the Properties tab, but their numeric name for all other purposes.

When editing the name of a component in the Properties tab, a numeric value is not saved as the name of the component.  
If this is to prevent components being given numeric names, a message should show when the user attempts to name their component a numeric value both when creating the component or when editing its properties. If a user is allowed to give components numeric names, then the given numeric name should show in the Properties tab and the name the user gives the component in the Properties tab should be saved.

tomcl, 2022-03-08:
OK, the problem here is pure numeric names which should be disallowed.
Quick to fix but not high priority so will do after Term ends

tomcl, 2022-04-16:
Closed for now, since it should be solved. Component names are due to be revisited some time for a better solution.

closed 2022-04-16 by tomcl