tomcl/issie #146: Wires do not correctly attach to import ports on components when reopening file from older ISSIE version

blayney, 2022-04-08:
V2.5.0, Windows 10

[image]

You can see in this screenshot that the wire previously connected to ALUF is now floating - the only workaround appears to be moving the end target of the wire (ALUF) which resyncs the end position of the wire. On big sheets opened for the first time on 2.5.0 this means moving every single component with missing input wires, which can be lots!
There's another missing input wire on a merge towards the top right. This issue persists across styling, and I'm not certain it only occurs on 2.5.0. If helpful, I can find the ISSIE version the files were created on. Thanks!

tomcl, 2022-04-10:
I will fold a fix for this into the next alpha release (soon).

tomcl, 2022-04-10:
OK - it looks like this was mended around 2.5.0a2 or so.

Download the latest 2.5.0a binary (we are now on a8). It will have bugs mended and extra features.

There will be proper releases when we are out of the alpha testing stage - till then you have to check for the publish date of release tags (they all have the same Issie version).

Let me know if this does not resolve the problem!

closed 2022-04-12 by tomcl