tomcl/issie #164: Unable to run two dev processes

jzzheng22, 2022-05-16:
**Describe the bug**
While developing ISSIE, I frequently run two `npm run dev` processes: one in a directory containing the master version of issie (i.e. this repo's master branch), and one in a separate directory containing my own issie fork. The new build system means that this is no longer possible.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `npm run dev` in one ISSIE directory
2. Run `npm run dev` in the second ISSIE directory
3. Error:

```
> issie@3.0.0beta.4 start
> cross-env NODE_ENV=development node scripts/start.js

path\to\issie\node_modules\webpack-dev-server\lib\Server.js:24
        throw error;
        ^

Error: listen EADDRINUSE: address already in use 127.0.0.1:9000
    at Server.setupListenHandle [as _listen2] (node:net:1372:16)
    at listenInCluster (node:net:1420:12)
    at GetAddrInfoReqWrap.doListen [as callback] (node:net:1559:7)
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:73:8) {
  code: 'EADDRINUSE',
  errno: -4091,
  syscall: 'listen',
  address: '127.0.0.1',
  port: 9000
}
```

**Expected behavior**
Ideally, one should be able to run ISSIE in dev mode in multiple directories without issue.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Node v16.15.0

jzzheng22, 2022-05-16:
CC @apantelopoulos

apantelopoulos, 2022-05-16:
Until a proper solution to this is found you can manually "hack" your 2nd directory by changing:

- ~~port: 9000~~ -> port: 9001 on file scripts/start.js (line 27)
- ~~sprintf $"http://localhost:9000"~~ -> sprintf $"http://localhost:9001" on file src/Main/Main.fs (line 133)

I cannot test it at the moment but this should work.

Let me know if it still fails to compile.

jzzheng22, 2022-05-16:
> Until a proper solution to this is found you can manually "hack" your 2nd directory by changing:
>
> * ~port: 9000~ -> port: 9001 on file scripts/start.js (line 27)
> * ~sprintf $"http://localhost:9000"~ [...]

Thank you! Can confirm that the workaround does work :)

jzzheng22, 2022-05-16:
Suggestion: can we set port using an environment variable?

apantelopoulos, 2022-05-16:
That's what I had in mind as well (with 9000 being the default if no port given). I will work on it in the following days

tomcl, 2022-09-16:
Closed - since it is not a typical use case and difficult to deal with nicely.

closed 2022-09-16 by tomcl