tomcl/issie #170: Dividerbar for WaveSim tab does not span the whole height when the right-tab section becomes scrollable

adidesh20, 2022-05-25:
**Describe the bug**
The right tab section for the Waveform Simulator is set to become scrollable when it overflows (i.e. there are a large number of possible waves). When scrolling down the list, the height of draggable dividerbar does not extend, instead staying at the original height of the section

**To Reproduce**
Steps to reproduce the behavior:
1. Open a sheet with a large number of sequential components (I used eep1 from lab5eep1model)
2. Click on the Waveforms button
3. Scroll down the list of waveform checkboxes
4. See how the the height of the dividerbar does not change

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Additional notes**
Discovered this issue on my fork, where long truth tables cause the same problem.

adidesh20, 2022-05-27:
Have now opened a PR that fixes this issue

jzzheng22, 2022-06-07:
#171  is not a perfect solution: the very first time it is loaded, the divider bar is still the wrong height. Any time after that it is fine.

I'm not really sure what the best way of resolving this is. Could perhaps force a reload if it is the first time the bar is loaded, but that would require mutable state I think? Maybe not worth it.

closed 2022-09-16 by tomcl