tomcl/issie #211: Automatic Deletion of Custom Components

apantelopoulos, 2022-09-01:
**Describe the bug**
Currently, when a sheet is deleted, Custom Components on other sheets originating from the deleted sheet are not automatically deleted. This could cause problems in the simulation if they are not deleted manually by the user. 

**To Reproduce**
Steps to reproduce the behavior:
1. Add a custom component to a sheet
2. Delete the schematic of that custom component 

**Expected behavior**
Such components should be deleted automatically.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version v3.0.6

tomcl, 2022-09-06:
A less invasive solution (perhaps better) would be to make the existing canvas checks done before constructing SimulationGraph check for this. They check for a lot of other things, including dependencies (subsheets) that have errors.

Simulator.runCanvasStateChecksAndBuildGraph

apantelopoulos, 2022-09-08:
Closed with #210

closed 2022-09-08 by apantelopoulos