tomcl/issie #220: Can’t simulate counter in step or wave mode when ‘LOAD’ input is derived from ‘Q’ output

edstott, 2022-09-16:
**Describe the bug**
Simulation commands are unresponsive when the design contains a counter with load, and the load signal is derived from the output of the counter. Simulation works correctly when load is driven by an input port. 

**To Reproduce**
Steps to reproduce the behavior:
1. Load attached design
[test.zip](https://github.com/tomcl/issie/files/9581967/test.zip)
2. Open Step Simulation tab
3. Click Start Simulation
4. Click 'Clock Tick 0'
5. At this point, further clicks of 'Clock Tick' and 'Goto Tick' have no effect. Wave simulation is similarly unresponsive

**Expected behavior**
Simulation should continue

**Desktop:**
 - OS: e.g. Windows 10
 - Version 3.0.0 built from https://github.com/tomcl/issie/commit/a83cc21920e497ee8b5c7527d268383ce3ed9827

apantelopoulos, 2022-09-16:
@edstott  Do you mean a design like this? Or something else?

[image]

edstott, 2022-09-16:
Like this:
[image]

edstott, 2022-09-16:
This is what you'd if you wanted a counter period that wasn't 2^n

apantelopoulos, 2022-09-16:
Fixed on #232

closed 2022-09-16 by apantelopoulos