tomcl/issie #222: Support more number formats in Bus Compare component

edstott, 2022-09-16:
**Is your feature request related to a problem? Please describe.**
Constants can be defined using multiple number formats. Bus Compare component only accepts decimal input. When the value is entered, the sheet shows it in hexadecimal format, which is inconsistent

**Describe the solution you'd like**
Allow the 'Compare With' parameter of Bus Compare to be specified in decimal (including signed), hex or binary

tomcl, 2022-09-16:
Yes! This was not done because doing it for constants was painful, requiring a lot of implementation changes as well as a component type change. The main pain is because it is necessary then to store the text creating the number as well as the number. The dialog functions were not initially created to do this.

tomcl, 2022-09-16:
The way to do this is:
(1) abstract the code currently implementing constant into one or more functions that will work for bus compare
(2) make sure constant still works
(3) implement bus compare similarly

PR #246 (2022-09-20): Support all number formats in Bus Compare component
Fix #222
closed 2022-09-20 by apantelopoulos