tomcl/issie #223: Verilog block doesn't allow non-constant shifts

edstott, 2022-09-16:
**Describe the bug**
The Verilog block reports an error if the right hand side of a shift operator is not a constant

**To Reproduce**
Steps to reproduce the behavior:
1. Enter this Verilog:
[image]
2. Component cannot be saved due to reported error

**Expected behavior**
The component should be accepted. This syntax is commonly used to make line decoders and shifters

apantelopoulos, 2022-09-16:
This requires the creation of an arbitrary width SHIFT component.

edstott, 2022-09-16:
Ah, so it's converting the Verilog into existing Issie components? That would explain the limitation

apantelopoulos, 2022-09-16:
Yes so that it would use existing code for step & waveform simulation, truth tables, etc.

tomcl, 2022-09-16:
We should  have a shift component. Parametrised by:

- input width
- output width
- shift width
- shift type (LSL, LSR, ASR)

apantelopoulos, 2022-09-20:
Closed with #247

closed 2022-09-20 by apantelopoulos