tomcl/issie #226: Add 'OK' button to 'Select Waves' Dialogue

edstott, 2022-09-16:
**Is your feature request related to a problem? Please describe.**
When setting up a Wave Simulation you need to choose the waves to display in a dialogue box. When you have finished, you must click the 'x' button to return to the main screen, which feels like a cancel action, rather than proceed.

[image]

**Describe the solution you'd like**
The box should have OK and Cancel buttons

tomcl, 2022-09-16:
Yes, I guess that would be clearer. It is conceptually correct because it is cancelling the wave selection box - but confusing

tomcl, 2022-09-16:
Compromise: box now has Done button.

https://github.com/tomcl/issie/commit/fa944236ecc9c0a02ff21a79b97f428febd5c7b2

To implement fully needs a bit more work: delaying all changes till the end (which would be mildly desirable anyway so that waveform creation is all done together).

edstott, 2022-09-20:
I think Done is fine. Thanks!

tomcl, 2022-09-20:
Great!. Closing.

closed 2022-09-20 by tomcl