tomcl/issie #227: Cursor switches to different field in parameter tab

edstott, 2022-09-16:
**Describe the bug**
When editing a parameter of a component (e.g. value of constant), highlighting the current value, as if to type a replacement, results in the cursor moving to the 'Component Name' field

**To Reproduce**
Steps to reproduce the behavior:
1. Place a component that contains a literal, like a constant
2. Select the component
3. Click and drag to highlight the numerical value
4. Type a replacement - your new value appears on the end of the component name instead of replacing the value you expected

**Expected behavior**
It should be easy to change literal values without deleting the existing digits one by one. It may be appropriate for the existing value to be highlighted by default when any numerical parameter field is selected, since it may be more common to type a complete replacement than change individual digits of the existing value

**Desktop (please complete the following information):**
- OS: Windows 10
- Version 3.0.0 built from https://github.com/tomcl/issie/commit/a83cc21920e497ee8b5c7527d268383ce3ed9827

edstott, 2022-09-16:
Actually, I can't repeat this bug now. Will update if I find the specific circumstances when it occurs

tomcl, 2022-09-20:
Keep this open for a while in the hope somone can reproduce it.

tomcl, 2022-10-08:
The focus CSS has been changed a bit in the properties window, which may help things. There is a fundamental problem that users often want to edit the diagram with properties window open, but also want to change properties on multiple symbols quickly. Having focus rules that fit all use cases is difficult.

The ambition at the moment is for focus never to lead to very surprising behaviour, and to make typical operations quick. Add to this conversation if you have a use case the current logic does not cope with,