tomcl/issie #229: Can't build Verilog for IceStick/IssieStick if it contains a 4-MUX or 8-MUX

edstott, 2022-09-16:
**Describe the bug**
My design wouldn't run the IceStick/IssieStick build process, with the console message 'Error in Verilog output: Cannot happen'. I found a minimal example which narrows the problem down to 4-MUX and 8-MUX. 2-MUX seems ok. The problem seems to be occurring before any of the external compilation tools are launched

**To Reproduce**
Steps to reproduce the behavior:
1. Load the attached test case: 
[test.zip](https://github.com/tomcl/issie/files/9583711/test.zip)
2. Enable the hardware build tab from the view menu
3. Select IceStick or IssieStick
4. Build process doesn't begin

**Expected behavior**
Hardware build should be possible with 4-MUX and 8-MUX

**Desktop (please complete the following information):**
- OS: Windows 10
- Version 3.0.0 built from https://github.com/tomcl/issie/commit/a83cc21920e497ee8b5c7527d268383ce3ed9827

apantelopoulos, 2022-09-16:
This is because of wrong Verilog Code generation for: MUX4, MUX8, DEMUX4, DEMUX8

apantelopoulos, 2022-09-16:
Fixed on #232

closed 2022-09-16 by apantelopoulos