tomcl/issie #251: Allow placement of copied user component without project reload

edstott, 2022-10-27:
**Is your feature request related to a problem? Please describe.**
It is common to reuse sheets from other projects by copying .dgm files between project directories.
However, if you do this when the target project is open, you need to close and reopen the project to access the copied sheet.

**Describe the solution you'd like**
Add an option under 'Sheets' to import a sheet from a different project.
That avoids users needing to copy their .dgm files at all, which can cause errors like move instead of copy.

**Describe alternatives you've considered**
The 'THIS PROJECT' section of the Catalogue could automatically update when .dgm files in the project directory are changed.

tomcl, 2022-11-05:
Good idea. To be added to minor changes when we next do dev.

tomcl, 2024-08-30:
this was closed by the very comprehensive sheet import function.

closed 2024-08-30 by tomcl