tomcl/issie #254: A way to rearrange the truth tables (they do not seem to come up in any systematic order as far as I'm aware)

saturn691, 2022-10-31:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

saturn691, 2022-10-31:
For example, when you add inputs A, B, C and outputs X and Y (in that order), the truth table could show B, A, C, Y, X

tomcl, 2022-11-05:
I am not sure quite what you mean by adding inputs and outputs in a given order?

The truth table logic takes part of a circuit and identifies all inputs and outputs. There is in general no obvious canonical ordering, so you are right that the order is rather random. It is correctly in order for MUX and DEMUX inputs and outputs.

It would arguably be worth allowing input and output order to be changed via drag-and-drop GUI, but this is quite a lot of dev work so not a high priority.

Perhaps I'm not understanding the problem?

edstott, 2022-11-05:
I think the problem is that it can be hard to compare a truth table to a design on paper if the ordering of rows is different.
Some possibilities
- A system for rearranging columns would be useful - a lot of work as you say
- Maybe an easier option would be an alphabetical ordering, so for example input 'A' is always left-most and always the MSB in the ordering of rows. Then at least it would be predictable.
- Could truth tables be selectable HTML tables? Then they could be copied and pasted into a spreadsheet and sorted.

tomcl, 2022-11-06:
OK - we had some old code that does this - in a not perfect way.

Please see v3.0.4 and comment

3.0.4 also has ability to fill in internal connections, so that a partial circuit defined by selecting components only behaves as expected.

The truth tables should be improved, e.g. a thick vertical divider between inputs and outputs. This unfortunately is difficult to implement given the existing code.