tomcl/issie #255: A better manager for JS packages

0x6770, 2022-11-03:
**Is your feature request related to a problem? Please describe.**
I had some error messages about "conflicting peer dependency" due to two versions of react being required in different scopes when I ran `bash build.sh` this afternoon. Although this problem has already been solved in a recent PR by manually fixed version number for react, it can be laborious to solve these conflicts manually every time.

**Describe the solution you'd like**
I tried using [yarn](https://yarnpkg.com) (classic/v1) instead of npm and rerun `bash build.sh`, the build process passed without the above error. 
Yarn is known for its better performance compare to npm:  parallel installation, better method to solve dependency. It could reduce future pkg conflicts.

**Describe alternatives you've considered**
Another popular replacement for npm is [pnpm](https://pnpm.io), however, it is **not supported** by Fake. 

**Additional context**
- This change requires only very little work, just search and replace npm with yarn in all build scripts. 
- A very informative blog comparing the three PM mentioned above, https://blog.logrocket.com/javascript-package-managers-compared/

0x6770, 2022-11-03:
`yarn` can be installed by `npm` (recommended) or package managers for your OS, e.g. `pacman`, `brew`

```bash
npm install --global yarn
```

 [official guide to install yarn](https://classic.yarnpkg.com/lang/en/docs/install)

0x6770, 2022-11-03:
Tested `yarn dist` and `yarn dev` works.

Environment:
```bash
uname -rmo
5.15.74-3-MANJARO x86_64 GNU/Linux

dotnet --version
6.0.402

node --version
16.18.0

yarn --version
1.22.19
```

tomcl, 2022-11-05:
This (I think)  is not quite correct. the peer dependency conflict here has non-overlapping ranges of react. The fix was to install a later version of simple-react-code-editor that removes the non-overlapping issue.

To check that the above is correct please run `yarn dist` on the "fixed by yarn" version and report back whether this runs ok without the recent version fix. I suspect that it will not run - but if it does I agree yarn is dealing with these conflicts better (by allowing two different versions of a package to co-exist?)

tomcl, 2022-11-05:
I've regressed your branch to the editor component v1.11 - try installing all packages compatible with that and see then if dist works. (In don't want to install yarn on my system unless we make this change).

0x6770, 2022-11-06:
Oh, I forgot to mention, my previous tests were run on 734a4b4e.

I've rerun all tests on the head of 255-a-better-manager-for-js-packages with `"react-simple-code-editor": "^0.11"`. 
`bash build` and `yarn dist` works.

A complete log can be viewed here https://asciinema.org/a/535288

tomcl, 2022-11-08:
So: I have done some checking. In fact npm works fine now with `^0.11`. Although yarn is slightly nicer UI, the added complexity of installing it does not seem enough of an advantage over npm given similar dependency management. Its main advantage now seem to be that it is faster.

tomcl, 2022-11-08:
I am closing this but it can be reopened again if more evidence for yarn being better than npm emerges.

closed 2022-11-08 by tomcl