tomcl/issie #256: 32-bit constants always return zero?

edstott, 2022-11-04:
**Describe the bug**
A 32-bit constant seems to always resolve as zero in simulations

**To Reproduce**
Steps to reproduce the behavior:
1. Place a constant and an output, both 32 bits wide
2. Set the constant value to non-zero
3. Run Step or Truth Table simulations
4. The output is zero

**Expected behavior**
32-bit, non-zero constants should be possible

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - v3.0.2

**Additional context**
I have not tried every width up to 64, but I have tried a few and not found another width that does not work

edstott, 2022-11-04:
32-bit bus selects also seem to be affected
Possible workaround is to use 2x16-bit constants or bus selects and a mergewires. Mergewires seem ok.

tomcl, 2022-11-05:
closed by release https://github.com/tomcl/issie/tree/v3.0.3

closed 2022-11-05 by tomcl

edstott, 2022-11-08:
Now it seems broken for any width!

[image]

I'm also seeing a problem if I try with splitwire instead:

[image]

v3.0.4 experimental