tomcl/issie #259: Crash when simulating wide datapaths

edstott, 2022-11-06:
**Describe the bug**
Issie crashes when simulating certain designs with larger datapaths. The exact circumstances vary with the design - sometimes it is on simulation start, sometimes on a certain tick number. Wave simulation and step simulation can crash on different ticks.

The bug does seem to be associated with operations on wide datapaths, because the example below is not complex in terms of number of wires, components or hierarchy depth. An 8-bit version of the design does not cause the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Load design below
2. Start wave simulation
3. Select output port for display
4. Advance simulation until crash (if no crash already)

**Expected behavior**
The example here is from a 32 bit shift-and-add multiplier, which seems like a reasonable use-case.

If there is a practical limit to bit widths then it may help to formalise maximum widths. Currently, the UI enforces a 64 bit limit for constants but not other components.

**Screenshots**
Core of a 32-bit shift-and-add multiplier
[image]

Crash when advancing to tick 16
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 3.0.4

**Additional context**
[test.zip](https://github.com/tomcl/issie/files/9946380/test.zip)

tomcl, 2022-11-07:
Thanks for this.

The only intrinsic limit is 64 bits for memory address or data.
There are UI limits for properties.
Unfortunately the simulation code (which prints data) has some conversion bugs for larger data (anything over 32 bits is stored as a bigint but then converted to 32 or 64 bit int for other processing).
I will do some bug fixing.
It is really just that the low-level "conversion to printable string" code has never been properly written for arbitrary widths.

tomcl, 2022-11-08:
OK - I have done some fixing. Unfortunately there is no simple way to make widths > 64 bits always work properly when displayed - they take up too much room. Try the experimental win binary under 3.04 release for the fixed version and let me know. I did a bit of testing and all seemed good for 128 bits.

edstott, 2022-11-08:
Got a problem with v3.0.4experimental1:

[image]

This simulation didn't crash with v3.0.4experimental, but the results were wrong due to #260. Crash occurred then clicking 'Start Simulation' in waveform viewer

tomcl, 2022-11-08:
#260 is now fixed??

Re long numbers. I agree I think mouseover is best solution, and ... to indicate MSB truncation: `....1234`, `0x...2a57`. LSB truncation is of course better for numeric values. Perhaps truncate MSB in binary/hex and LSB in decimal?

Mouseover is a bit tricky, but I can do MSB/LSB truncation easily - I will see if it can be added.

Scientific notation would be a good alternative to decimal, I agree.

We should have the simulation reworked for faster and much longer simulations. After that, waveform simulation can be allowed from any starting point (like step simulation). In conjunction with that high bit-width display and also input can be considered. Input is less important, but constants and bus compare should at least be up to 32 bit.

tomcl, 2022-11-08:
> Got a problem with v3.0.4experimental1:
> 
> [image]
> 
> This simulation didn't crash with v3.0.4experimental, but the results were wrong due to #260. Crash occurred then clicking 'Start Simulation' in waveform viewer

Weird - I tested it with 33 bit in, 32 bits selected and it worked fine. I will try some more.

edstott, 2022-11-08:
> Weird - I tested it with 33 bit in, 32 bits selected and it worked fine. I will try some more.

This one has a 64 bit register and output

tomcl, 2022-11-08:
Were you perhaps using an older version?

edstott, 2022-11-08:
[fract.zip](https://github.com/tomcl/issie/files/9960940/fract.zip)

Here's the project. Sheet `m32nstest` is the failing simulation. I'll try to find a minimal example

tomcl, 2022-11-08:
OK, experimental2 should fix it!

tomcl, 2022-11-08:
> #260 is now fixed??
>
> Re long numbers. I agree I think mouseover is best solution, and ... to indicate MSB truncation: `....1234`, `0x...2a57`. LSB truncation is of course better for numeric values. [...]

Try out `experimental3` for better display of wide numbers.

I have now implemented better display of large numbers. I do not do scientific notation, but I do MSB truncation flexibly as follows:

(1) if possible display as normal, decimal, or hex, binary with leading zeros included
(2) For decimal, if that is too wide, display as `..lsDigits`. lsDigits is digits with MS trunxated as needed to fit
(3) for hex, binary, try to display exactly, but removing leading zeros, in form `w'xNonZeroDigits` or `w'bNonZeroDigits` where `w` is the correct width in bits,
(4) If that is too large display truncate in the form `w`..lsDigits`

This is close to verilog syntax but not identical. I want to keep `x` consistent with programming hex literals I think.

tomcl, 2022-11-08:
Ok, `experimental4` is as `experimental3` but with better (?) positioning of text on waveforms.

edstott, 2022-11-08:
Looks good to me. Stability is the main thing.

Truncation is only in the right hand column and not the waveform itself, is that right? The right hand column could be variable width to accommodate long values.

I'm not sure the `w'` is essential because the width can be seen in the left hand column anyway

tomcl, 2022-11-18:
The RH values column width could do with some fine tuning - for a later release. It _could_ be made responsive to the max width displayed (in current radix). Not however a priority.

closed 2022-11-18 by tomcl