tomcl/issie #260: 32-bit bus selects always return zero

edstott, 2022-11-07:
**Describe the bug**
A 32-bit bus-select always resolve as zero in simulations

**To Reproduce**
Steps to reproduce the behavior:
1. Place a 33-bit constant constant and select 32 bits of it to pass to an output.
2. Set the constant value to non-zero
3. Run Step or Truth Table simulations
4. The output is zero

**Expected behavior**
32-bit bus selects should be possible

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - v3.0.4

**Additional context**
Same problem as #256?

tomcl, 2022-11-08:
This was actually a different issue - the > 32 bit logic has not been well tested, and a different set of low-level functions were just wrong for > 32 bit inputs.

v3.0.4 experimental now has a fix for this and #259 . Let me know if it works ok (it did in my testing).

edstott, 2022-11-08:
Now it seems broken for any width!

[image]

I'm also seeing a problem if I try with splitwire instead:

[image]

v3.0.4 experimental

tomcl, 2022-11-08:
Now fixed experimental1. I hope.

(sorry - it was a typo in my bit selection function introduced when mending other parts of it - had it selecting n-1 bits)

tomcl, 2022-11-08:
This issue should be fixed now (seems to be) by experimental1 along with #259

tomcl, 2022-11-08:
Currently the experimehtal version has a "Help!" message printed out. I am leaving that for the moment. Everything works fine when I test it. Basically, bigints are allowed to overflow in the simulation - only the LS bits are taken. However that is a bad practice - I need to be sure that I am masking out bits whenever they matter. The thing is, masking on input and output is not efficient. I should probably change it so that bigints are consistently masked based on output width and I'm not sure why that is not done ATM. I will have a look, because bigint masking is separate from <= 32 bit logic so there could be some extra bugs in there.

tomcl, 2022-11-18:
I'm closing this as fixed by v3.0.5

closed 2022-11-18 by tomcl