tomcl/issie #283: Wrong bitwidth when using N bit NOT

petraratkai, 2023-05-15:
**Describe the bug**
When connecting an output port to an N bit NOT component, the values displayed have more bits than the output port.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a 5 bit input port, a 5 bit NOT gate, 5 bit output port
2. Simulate

**Expected behavior**
Display value with the expected width

**Screenshots**
[image]

petraratkai, 2023-06-10:
The following circuit also produces an error:
[image]
C1 is 1, the output of the inverter should be 0, compared to 0 should output 1 but Issie outputs 0.

lucasng32, 2023-08-02:
Could I be assigned to this?

tomcl, 2023-08-02:
Done.

lucasng32, 2023-08-03:
The bug still exists on the new simulator
A bigint version of NOT exists for NOT on busses > 32 bits: 
```
| NbitsNot numberOfBits, true ->
        let a = insBigInt 0
        // failwithf $"TODO: fable does not support op_OnesComplement function"
        // BigWord (System.Numerics.BigInteger.op_OnesComplement a)  FIX: 2^n-1-a
        let w = comp.InputWidth 0
        // (bigint^w)
        let (minusOne: bigint) = ((bigint 2) <<< w) - (bigint 1)
        let res = minusOne - a
        putBigInt 0 res
```
I implemented NOT for <= 32 bits similarly. I noticed the bitshift seemed to be circular so I added a case for 32 bits: 
```
| NbitsNot numberOfBits, false ->
        let a = insUInt32 0
        let w = comp.InputWidth 0
        let minusOne = (1u <<< w) - 1u
        let res = 
            match w with 
            | 32 -> minusOne - a - 1u
            | _ -> minusOne - a
        putUInt32 0 res
```
It's also my first time contributing, I've read through the wiki and wanted to clarify some things: 
I've created a fork of master, created a new branch, applied the changes and then commit. 
I then pushed the branch to my fork. 
Is the next step to open a pull request to this repository's master branch?

tomcl, 2023-08-03:
Assuming that you have rebased on master (and dealt with any conflicts due to its change, if it has changed) yes,

tomcl, 2023-08-04:
closed by #342

closed 2023-08-04 by tomcl