tomcl/issie #303: Upgrade Issie gates to be N-input

tomcl, 2023-07-14:
A common request last year was to allow 3,4 etc input gates.

This has four parts:

- Rationalise all 2-input gates to implement them with a single new ComponentType case `Gate` with fields to select function. This will make many parts of Issie easier since gates are nearly always all treated identically.
- Upgrade ComponentType Gate definition to add a "number of inputs" field
- Add input number Change to properties and input number selection screen to Catalog
- Use view function code from [Team 5 branch](https://github.com/tomcl/issie/tree/team5-23-gates).

To preserve backwards-compatibility with old designs use #302 and convert old AND/OR/etc components into new-style Gate components with 2 inputs.

tomcl, 2023-08-17:
This looks a pretty good uptodate description. Just to say:

- To stop it getting silly should have a limit to number of inputs.
- For rectangular gates this could be high (e.g. 16)
- For distinctive-style gates it is more challenging and maybe the practical limit would be 4.
- I suggest, when implementing rectangular gates, make the limit a configurable constant initially 4.

The most interesting part of the implementation is the simulation:

FastCreate - add code that creates variable length input IOArray accoding to number of inputs.
FastReduce  - add code that implements the gate function with a variable number of inputs
Truthtables - need to implement constraint reduction and algebraic simulation (more challenging)

ccrownhill, 2023-08-19:
I have started implementing an `AndN` component and managed to get the step simulation to work.

However, for the truth table, I have to modify `fastReduceFData`.
I have started to try and understand how this function works based on `FData` but I have not yet succeeded to completely understand how it works in detail.
I will also try to read Yujies dissertation but in case I still don't understand this function afterwards, I was wondering whether
you could explain how `fastReduceFData` works a little more?

tomcl, 2023-08-19:
I’ve pushed a commit to master with more doc on that function, and also the crucial IOArray type.

FData allows algebraic data in which a component input or output its implemented as a
Mathematical expression of the inputs. E.g. an AND gate with inputs a, b would be have output a AND b.
Note that these expressions can be arbitrarily large trees.

Not sure Yujie’s report will help much, though it will have some context and a detailed description of
IOArray.

tomcl, 2023-08-19:
And one more commit with detailed XML documentation of the FastComponent type.

Also read non-XML doc of FastSimulation type (maybe this should be converted to XML doc?)

Tom

ccrownhill, 2023-08-21:
Should I delete the old gate components in CommonTypes.ComponentTypes and keep them only in the module JSONComponent or keep them in both?

tomcl, 2023-08-21:
I'd keep them in until SURE everything works properly. For example you can test new vs old.

But after that, their matches can be removed and a conversion done on loading old sheets.

ccrownhill, 2023-08-21:
Ok, I will do that then.

ccrownhill, 2023-08-22:
Should I also create N input versions of the N bit gates?

ccrownhill, 2023-08-22:
One question about the right tab of Issie.
I have added a dropdown to change the gate type.
However, I have manually set a magic value height so that it doesn't overflow the bottom of the properties tab.
From my analysis I saw that the height depends on the amount of content in the properties tab.
Do you think there is any way to avoid this magic value and find out the maximum height in that case?

ccrownhill, 2023-08-22:
Regarding the catalogue menu: I was thinking to keep a separate item for every gate so that it requires less clicks to create a gate (otherwise, the user has to click on gate and then specify its type in the popup).
In the same line of thought I was considering to leave menu entries like And2, Or2, etc. for 2 input gates.
What do you think of each of these ideas?

tomcl, 2023-08-22:
Yes, remember you can use web zoom on the whole Ui which changes available space.

You can work out the height of elements, or of the RH tab etc, and use that to adjust things. It is a bit involved but possible. One thing is that maybe dom element has to be  has to be stored in an option mutable variable (see sheet which does this) since it is not available till after the view function has run – when the dom exists. So then after dom has been created the height will be correct – which is fine.

tomcl, 2023-08-22:
My suggestion:

Keep gates as now.

Default to 2-input

Add number of inputs to properties, so you can change it there.

Somehow, add a hint to catalog so it is obvious that is how you make n input gates.

You can use the HintPane maybe while dragging a gate onto canvas to say how to make it multiple input?

ccrownhill, 2023-08-22:
I have implemented these things now.
Regarding the Dom element I did not wrap it inside an option mutable variable since it is also used in lots of other functions and I thought that I saw that these also use it directly to do `document.getElementById`.
Since it worked I have kept it that way.

I have made a pull request with this: #363 

What do you think of this implementation?
I tested it with some term 2 projects.

Next I would go on to implement the old style gates.

tomcl, 2023-08-22:
document.getElementById is fine if it works - the issue is only first time view function runs can give error, hence mutable option etc. maybe not an issue? I remember it being a pain in some cases.

I will check implementation and do PR probably this evening.

tomcl, 2023-08-23:
Re old style gates. I've worked out how you can do this compatible with rect gates for any number of inputs. 

1) use a gate symbol which can be stretched a bit and will manage 2 or 3 inputs
2) for more inputs use the same symbol (with 2 or 3 inputs) but add a vertical line extension of the (AND/NAND) gate input edge line equally in both directions for extra pairs of inputs. This will correspond to the input-side edge of the rect gate.  Do the same thing with the curly gates but keep the curly part of the input edge as it would be for 2 or 3 inputs.

See comments here for some example pics https://github.com/circuitikz/circuitikz/issues/383

Doing this you can I think keep input port positions identical as you switch - except that the curly gates cause a small movement. It is very difficult to deal with this - I think you just accept that some wires will need rerouting for input NOR/OR/XOR.XNOR gates. it is a tiny reroute so should not change things. In any case we do not expect symbol types to be changed a lot.

tomcl, 2023-09-03:
closed by #375

closed 2023-09-10 by tomcl