tomcl/issie #307: Add-a-sheet feature

tomcl, 2023-07-14:
Currently it is possible to add a sheet to Issie by extracting a file manually from one project and copying it to another.

It would be good to have an "add sheet" menu option with the following functionality:

Check that a project is open (otherwise will not work)

- Use electron file selector to select a dgm file not in the current project to add (enforce that it is not in current project)
- Check that file name is distinct from the name of any file in current project - if not append "1" to it.
- Copy file to current project directory.
- Confirm op with a nice popup that details what has been aded, how many components etc.

Nice to have: allow multiple file selection and copy.

IlanIwumbwe, 2023-07-21:
May I please be assgined to this issue?

tomcl, 2023-07-21:
Done!

IlanIwumbwe, 2023-07-22:
I've decided to put the button for this under the sheets dropdown like so: 
[image]
I also named it import sheet, which I used as the type name for the UICommandType D.U, as AddSheet already exists. 
At the moment, it checks whether the project is open, and prints a message in the console. 

May you please help me with some hints on how to use the electron file selector to import the file? I've been looking through the code used to open projects to see if I can start from there but I haven't been successful. 

Thank you.

tomcl, 2023-07-22:
Agreed re the UI.

You can read the [dialog  electron API](https://www.electronjs.org/docs/latest/api/dialog) for the file selector. The open project code is close to what you want - it looks for a .dprj file. You will need to create your own function that uses the file open dialog.

The open file dialog can be used via the F# electron API interface (common/ElectronAPI)    showOpenDialogSync   

The interface tells you the possible parameters of this function and the comments what it does. Compare with existing uses in issie for examples (the interface JS compatible types a re a bit weird). We should proibably have a wiki page on using electron API functions.

have a look and ask further.

tomcl, 2023-07-22:
Re UI commandtype D.U. UI commands are just badly implemented at the moment. they should all be rationalised with the correct messages so there is only one Issie model update per UI command. then the UI command start and stop messages would not be needed. Why are they used? The concern is that a long UI command with multiple messages could get interrupted in the middle by another UI command, which would cause chaos.

Issie UI functions badly need a complete rewrite!

IlanIwumbwe, 2023-07-22:
Alright, I'll read the dialog electron API and look through Issie for existing examples.
Thanks

tomcl, 2023-07-22:
```
/// Ask the user to choose a project file, with a dialog window.
/// Return the folder containing the chosen project file.
/// Return None if the user exits withouth selecting a path.
let askForExistingProjectPath (defaultPath: string option) : string option =
    let options = createEmpty<OpenDialogSyncOptions>
    options.filters <- Some (projectFileFilters |> ResizeArray)
    options.defaultPath <-
        defaultPath
        |> Option.defaultValue (electronRemote.app.getPath ElectronAPI.Electron.AppGetPath.Documents)
        |> Some
    let w = electronRemote.getCurrentWindow()
    electronRemote.dialog.showOpenDialogSync(w,options)
    |> Option.bind (
        Seq.toList
        >> function
        | [] -> Option.None
        | p :: _ -> Some <| dirName p
```

this function in FilesIO opens a dialog to let the user choose a `.dprj` file. You need to make a new set of options like `projectFileFilters` to search for .dgm files. 

```
let private projectFileFilters =
    createObj !![
        "name" ==> "ISSIE project file"
        "extensions" ==> ResizeArray [ "dprj" ]
    ] 
    |> unbox<FileFilter> 
    |> Array.singleton
```

Probably there should be a single  `makeFileFilter (extn: string) (name: string)` function called to implement different types of dialog. Write this and then replace `projectFileFilters` references by a call to this if you want to write **clean code**, otehr wise juts cut and patse this function with a new name an slightly different definition of you want **unfunctional code with lots of unnecessary replication**.

You could change the options for multiple file selection. The dialog will then return an array - I am not entirely sure what F# type but it is pretty easy to use the IDE to try things, seeing what types it gives you.

IlanIwumbwe, 2023-07-22:
I see. This is a good starting point, I'll start from here and get back to you with any queries.

tomcl, 2023-07-22:
https://github.com/tomcl/issie/wiki/Using-the-Electron-API-(or-other-JS-library-code)-in-Issie

IlanIwumbwe, 2023-07-22:
> https://github.com/tomcl/issie/wiki/Electron-API-(or-other-JS-library)-interface-in-Issie

Thanks

IlanIwumbwe, 2023-07-27:
**Update on feature progress and other changes:**
- Sheets can be imported in within ISSIE and used as blocks in other sheets. 
- Multiple sheet selection is possible
- When sheet is in the project directory, it doesn't get imported
- Project must be open for sheet to be imported

At the moment, I use this function from FileIO.fs to move a file after creating a new path for the sheet:
`
let rename (oldPath: string) (newPath: string) =
    #if FABLE_COMPILER
    fs.renameSync (oldPath, newPath) // Asynchronous.
    #else
    File.Move(oldPath, newPath, false)
    #endif
`
Apparently the function 'copyFileSync' should be able to copy the files and not move them, which is what this does. I tried writing a new function to do this, but I think I'd have to add the 'copyFileSync' function to IExports.fsi. 
I think there's something I'm missing though about how I can do this another way.

To do:
- Add popup to show info about loaded sheet(s)
- Copy the files not move them

tomcl, 2023-07-27:
For copy file, you have read file and write file. Use them both to implement copy! Node does not have a copy file function.

Finer points:

- make sure the sheet name obeys the same constraints we have on names when a new sheet is created
- if sheet is in project directory use a choice popup with options to overwrite the existing sheet or add it as a sheet with a new name (like windows explorer does!). Look at references to UI.popupView.choicePopup for examples.

IlanIwumbwe, 2023-07-27:
Right.

Thank you very much

tomcl, 2023-07-28:
**Feature Request**

We have one problem with add-a-sheet. It can create malformed projects which will break the rest of Issie if it changes dependencies. Cases:

- A sheet added is dependent on another sheet which does not exist (same name and signature)
    - it is ok if such a sheet does exist either in the set of added sheets, or in Issie
    - be careful of corner cases here, e.g. if dependency exists ok in added sheets, not in current project, but then the dependency is renamed.
- Replace a sheet which is a dependency of another, however the signature of the replacement sheet is different

The question is what to do in these cases:
- For replacing a sheet which is already used as a dependency, with an incompatible sheet, we already have code which can match up the new and old signature and change all references to match the new signature. ideally this could be used (optionally). More simply that could simply not be allowed.
- When a dependency of an added sheet is missing:
     - you could look for the dependency on the same directory (it should be there) and if it is offer to add it as well.

The simple option is to refuse to do anything unless everything checks out perfectly (and that might be the best bet initially). But as you can see above better options are available. What cannot happen is to allow an Issie project which has an inconsistent state after an import.

For code working with custom component dependencies (you may be able to reuse some functions) see `customCompPorts.fs`. It is quite dense code - I wrote it.

For review @IlanIwumbwe

IlanIwumbwe, 2023-07-29:
I see. I agree that sorting out these dependency issues on a case by case basis is a better option.

I could implement something where before importing a sheet, I check in the destination directory for dependencies. For any non-existent ones, I look for them in the source directory and offer to add those as well.

I may be misunderstanding this, but when sorting out replacement of sheets that are dependencies of others, won't we have to not only change all signature references, but also check that the replacement sheet is physically compatible with the components the user wants to connect to it in some other sheet? Since the user may build some circuit for instance a 16 bit register with 16 D-flipflops, but name it 'fulladder'. Then when they try to import it, they are asked to overwrite or rename, then they overwrite an actual full-adder with something that isn't.

tomcl, 2023-07-29:
There is a use case for overwriting sheets with dependencies, or used as a dependency. All you can do is check that the sheet does not break Issie (signature is correct, dependencies exist). You cannot guess whether sheet has same function, or indeed whether user wants to change its function.

There is a (less strong) use case for overwriting a sheet with one that has a slightly different signature. Since we have code that copes with changing a sheets signature nicely (customCompPorts) that same code could deal with a changed signature – keeping same ports connected on instances, leaving new ports for manual connection, etc.

My suggestion is:

  1.  Check the set of imports for dependency issues (maybe initially just throw an error when any exist)
  2.  Throw up a confirmation saying what is being imported and its status (e.g. are there dependencies, are they in imported files, etc) This screen can explain anything needed. This screen has an OK or cancel response. If the import is impossible to make safe (however you decide to allow this) the screen can say this and its ok button will be greyed out.
  3.  If user decisions are needed on files, add these as tick boxes (see wavesimselect) or buttons. A set of rows (maybe a table)  one per file showing status and options for each.

This is quite a lot of work so do it incrementally – and stop when you have had enough. The first option would be to disallow anything that requires user decision. The next to allow user-confirmed replacement as long as there are no dependency issues. Etc.

One of the important UI issues is that being asked to make decisions one-by-one without full context is a bad idea – so a screen which analyses everything and provides the user info is a good idea. Look at the existing conformation when a custom component sheet has its interface changed for an example of this sort of thing (although in that case there are only two user options – change instances to be compatible or leave them).

We are trying with Issie to take complex requirements and expose them with a UI that makes all user decisions simple, and also makes it impossible for users to do bad things!

In writing this code do remember to check customCompPorts and its helpers for functions you can re-use relating to signatures etc.

Best wishes, Tom

IlanIwumbwe, 2023-07-29:
Right. I understand these suggestions. 

Thank you very much

IlanIwumbwe, 2023-08-01:
About the issue of having to show multiple popups for each file that causes import conflicts, what do you think about the idea of checking each path beforehand, working out which ones need user choice, then having one popup with a row for each file, with 2 buttons, 'rename', 'overwrite'. And a button at the bottom that is always greyed out until a decision is made about all files?

tomcl, 2023-08-01:
Agreed, need a bit more than this.

For each imported file you need to display info in one row (not all of, but as needed to make decision)):

  1.  Is it existing file? Does it have same signature as existing file?
  2.  If so, does it have existing or other imported sheets dependent on it
  3.  Does it have dependencies? What are their names? For each one, is it in the imported files, or in the existing project (with correct signature)

Where an imported sheet has dependencies things are complicated: they could be imported with it, or existing, or both

Then the decision is for each file: overwrite, rename, cancel. If the imports are ordered with dependencies at top it will make sense. Info for other sheets may change when the import of dependencies is resolved.

Note that in most cases all info will be empty, the sheet will have no dependencies and be different from all others,  and there will be no buttons.

The decision at the bottom – to dismiss confirmation – is cancel or import – with import greyed out until all necessary decisions are made.

This e-mail does not quite answer the question of what is needed to do this nicely and simply. Maybe reduce options?

Tom

IlanIwumbwe, 2023-08-01:
I see, thank you very much. I'll probably start by refusing imports that cause complex dependency issues, then incrementally adding options as I go.

IlanIwumbwe, 2023-08-04:
I've been working on a function which receives a sheet that I'm trying to import, and returns info about:
1. Whether it already exists
2. All dependents of it in the directory to which I'm trying to import to if it already exists
3. All sheets in the source directory which are dependencies of the sheet I'm trying to import.

I'm stuck on the third point, but I'm thinking about writing a function that uses `loadAllComponentFiles` from FileIO to give me all loaded components from a given file. Then the idea would be to get all dependents for all sheets from the source directory except the one I'm trying to import, then check for the sheet I want to import. If it appears, then the sheet is a dependency, and should be offered to be imported as well. Looking through CustomCompPorts, there didn't seem to be a function that did that so I decided to write one that does it this way.

tomcl, 2023-08-04:
The set of all sheets in current project is:
Model -> CurrentProj -> LoadedComponents

You can work out all dependencies from this by extracting component list from CanvasState and looking for custom components.

However, this has already been done (mostly)

CustomCompPorts:
getDependents - finds all sheets dependent on a given sheet in the current model.
getDependencies - not there, but should be quick for you to write it given the getDependents code as example.

Notice the useful parseeSignature function.

IlanIwumbwe, 2023-08-04:
Thank you very much

tomcl, 2023-08-05:
OK, juts found something else - better for your purposes. (Issie is so large I had forgotten where it was, and could not lo0cated till I remembered the type name!). Issie helper functions need rationalising.

The code below delivers a map of subsheet trees keyed by sheet name.

fileMenuView:

type SheetTree = {
    Node: string
    Size: int
    SubSheets: SheetTree list
    }

/// get the subsheet tree for aa sheets
let getSheetTrees (p:Project) =
    let ldcMap = 
        p.LoadedComponents
        |> List.map (fun ldc -> ldc.Name,ldc)
        |> Map.ofList
    let rec subSheets (path: string list) (sheet: string) : SheetTree=
        let ldc = Map.tryFind sheet ldcMap
        match ldc with
        | None -> {Node=sheet; Size = 1;SubSheets = []}
        | Some ldc ->
            let comps,_ = ldc.CanvasState
            comps
            |> List.collect (fun comp -> 
                    match comp.Type with 
                    | Custom ct when not <| List.contains ct.Name path -> 
                        [subSheets (ct.Name :: path) ct.Name]
                    | _ -> 
                        [])
            |> (fun subs -> {
                Node=sheet; 
                Size = List.sumBy (fun sub -> sub.Size) subs + 1; 
                SubSheets= subs
                })
    p.LoadedComponents
    |> List.map (fun ldc ->ldc.Name, subSheets []  ldc.Name)
    |> Map.ofList

IlanIwumbwe, 2023-08-05:
Thank you very much!

IlanIwumbwe, 2023-08-08:
Good morning Dr Clarke. I'm having a road block on implementing the UI to show the user relevant information needed to sort out dependency issues. I think the code that shows the UI below could be a good example, but I can't find it in the code base. 
Additionally, when implementing the getDependencies function, since I want to get dependencies from the source directory, which isn't the current project in the model, how can I access the relevant project and use that instead? 

Thank you

tomcl, 2023-08-08:
https://github.com/tomcl/issie/blob/86d137ef002fb7e694fe950abd7691b33b03e775/src/Renderer/UI/CustomCompPorts.fs#L508C1-L599C48

See the customCompPorts popup or (simpler) the info popup:

https://github.com/tomcl/issie/blob/86d137ef002fb7e694fe950abd7691b33b03e775/src/Renderer/DrawBlock/PopupHelpers.fs#L775

Basicallyu – you need a function to create a popup the body of which is the react you want to display.

Then you need to add the popup to the popupViewFunc filed of Model (to bring it up):

https://github.com/tomcl/issie/blob/86d137ef002fb7e694fe950abd7691b33b03e775/src/Renderer/Model/ModelType.fs#L538C3-L539C9

Two ways to do this – using dispatch and a ShowPopup message or directly by returning a changed moel in update function.

To remove just set that field to None. Some of the popup builder functions have automatic ways to do that built in (but it is easy to do yourself).

popupViewFunc is called from MainView via viewPopupFunc. You can see from this how every time the view function is executed the poup function is executed, with parameters the latest Model -> popupDialogData. So you can set popupDialogData arbitrarily to provide inputs to the popup.

It is a bit weird but the popup cannot see model directly. This is probably wrong. For many cases you want what is displayed to depend on current model was well as (controlling the popup from when it was started) popupDialogData.

https://github.com/tomcl/issie/blob/86d137ef002fb7e694fe950abd7691b33b03e775/src/Renderer/DrawBlock/PopupHelpers.fs#L726C1-L738C1

So probably we have popup technical debt in not always calling popup functions with current model as well as popupDialogData – getting round the limitation has made some of the popup code quite painful.

Anyway that does not much affect you. Popups can change model because they are passed the dispatch function and can therefore make react buttons with Ob=nClikc functions that send messages.

There should be a wiki page on Popups. So if you feel enthusiastic you can write one, and also optionally change the popup function stack so that popups are called with the whole of model. But those things are long-term – to resolve your issue you just need to use a popup with a few buttons or checkboxes.

We can have a teams call if you have qs

Best wishes, Tom

IlanIwumbwe, 2023-08-08:
I see. Thank you very much for this, I will be in touch when more questions arise.

IlanIwumbwe, 2023-08-10:
**Update**

I've got a popup to come up like so for all sheets I import:
[image]

I need to complete the getDependencies function and add checkboxes for all such sheets so that the user can choose to import them as well. 

You aren't allowed to import unless some decision is made for all sheets you import. 

I wanted to change the text to show which decision the user has made by creating a type 
:
`type choiceAction =
    | RenameAction
    | OverwriteAction
`
Then updating a mutable variable with the action chosen (which can change before popup is closed / import clicked):
`        
        // mutable decisions map to keep track of decisions made per sheet
        let mutable decisions : Map<string, string> = Map.empty

        // mutable currAction to change popup text based on whether overwrite or rename is chose
        let mutable currAction : choiceAction Option = None

        // Function to update a decision for a sheet
        let updateDecision (sheetPath: string) (newSheetPath: string) (action : choiceAction)=
            decisions <- Map.add sheetPath newSheetPath decisions
            currAction <- Some action
`
I generate message with this function:
`
let generateMessage =
            match currAction with
            | Some RenameAction -> 
                // Create a message for rename action
                str "You have chosen to rename the imported sheet."
            | Some OverwriteAction -> 
                // Create a message for overwrite action
                str "You have chosen to overwrite the project sheet."
            | _ -> 
                // Default message or no action chosen yet
                str "Would you like to rename the imported sheet, or overwrite the project sheet?"
`

Then add that to the React element:
`
            div [] [
            p [] [ str description ]
            overwriteButton
            renameButton
            p [] [ generateMessage ]
            ]
`

But this doesn't work. There's probably a better way to do this, maybe change button colour? 
Do you suggest any other way of keeping track of all decisions other than using a mutable type?

Thank you very much

tomcl, 2023-08-10:
One thing most people don't understand is that mutable variables are bad. We use them very carefully in Issie. So no mutable variable. The lack of explicit sequencing of updates makes reasoning more difficult and even though in special cases it is easy, when there are bugs checking that there are no weird errors is a lot more difficult if you have mutables.

Without that - you need to think carefully about state. Where a popup needs state that can be contained in popupdialogdata - any of the fields there are free for your popup to use, and you can have need be add other ones. In your case you want to add a Map type (initialised to Map.empty). Rather than have a different type you could juts use bool and name the map field: sheetOverwriteDecision. No map element means decision not et made etc (so the map itself effectively gives you an option on its data).

You generatemessage function is a value not a function. You need to make it have a unit parameter (or be fun () ->)

Look for examples at otehr Issie code that uses buttons with onClick properties that call functions. Note that the function can have diapastch is parameter and tehrefore send a message to chnage state (in this case the popupdialogoptiuons state).

IlanIwumbwe, 2023-08-11:
I'd like to ask for help on how to deal with a certain issue.

I was testing adding information about sheet import, where if a sheet name already exists in destination directory, the user gets told whether or not that sheet has the same signature as its instances in all other sheets that depend on it. If it doesn't, then the overwrite button should be disabled, since otherwise we have conflicts. 

This is the function I use to get information about each sheet:

`// get relevant info about a sheet for display on popup
let getSheetInfo (model : Model) (oldSheetPath : string) (newSheetPath : string) =
    let sheetName = oldSheetPath |> baseNameWithoutExtension

    // get sheets in current project that would depend on an existent sheet, same as one that's being imported
    let exists, depSheets =

        if newSheetPath |> exists then 
            match getDependents model sheetName with
            | None -> true, ""
            | Some (newSig, instances) ->
                true,
                instances
                |> List.map (fun (sheet,_,sg) -> sheet + (string <| dependencyDoesNotMatchSignature newSig sg))
                |> List.distinct
                |> String.concat ","
        else false, ""

    exists, depSheets
`

I use the `dependencyDoesNotMatchSignature` function to pass the signature of the sheet I want to import, and the signature of the instance of that sheet in some other sheet. Used as it is there just for testing. I setup a test where I expected that the sheet I want to import has a different signature from its instance in the test sheet, but I still got _false_ output from the function.
[image]
_The normal and4 had no T1 output._

If I understand how the 'getDependents` function works, then I think the problem is that when I pass the sheetName, it is compared with all loaded components from the current directory, so the signature I get is of the component I've changed, not the component I'm trying to import. I'm not sure whether there's other ways to solve this, but maybe we can also pass the project we want to consider, so I can pass the project in which the sheet belongs and search its components. I'm not sure how I can access that though, maybe there's some function that gets the .dprj file and returns this project type? Or maybe I can use some function and pass the project directory to get the components and pass those in the  'getDependents` function directly.

Thank you very much

tomcl, 2023-08-11:
Solution here, as in many cases, is to think functionally and if need be add parameters.

You can use getDependents – you just need to do:

model
|> (change name of current sheet)
|> update current sheet contents with new sheet
|> getDependents

Using Optics it is quite convenient to do that.

Alternatively, and neater, rewrite getDependents as a function getDepenentsOfSheet with an extra parameter for the sheet to find dependents of, , which you can ue easily, and a new short definition of getDependents as now using your getDependentsOfSheet.

Does that help?

There should not be anything obscure about the code, it is all pure functional.

Tom

[image]<https://user-images.githubusercontent.com/56346800/260036717-a62850ff-edbb-412e-9c04-1c77d31d67d7.png>

IlanIwumbwe, 2023-08-11:
As you have suggested, the getDependents function above takes an extra paramter for the sheetName, I just rewrote it in FileMenu.fs. I will change that though, and put it in CustomCompPorts, then use it in the getDependents function that is already there, passing the current sheet from the currernt project as input. 

I think this still doesn't work for signature comparison since the loaded components are still from the current project from the model passed, yet I want to compare from components in the project from which the sheet is being imported. This is why I thought adding another parameter for the loaded components that should be searched might help.

tomcl, 2023-08-11:
Yes, although doing:

let model’ =
model
|> set loadedComponents_ newComponents

Is not a big cost instead!

IlanIwumbwe, 2023-08-11:
I see. Thank you very much

IlanIwumbwe, 2023-08-12:
Before submitting a PR, I wanted to get rid of the mutable type I was using to keep track of user decisions before import. I decided to add an ImportDecisions map to UserData in ModelType.fs. The method I've tried isn't working, the map doesn't seem to get updated with more than one file at a time. Sometimes the map doesn't seem to get updated at all. 

I have a function that I call each time Overwrite or Rename is clicked:
`       
let onButtonClick (sheetPath: string) (decisionOption: ImportDecision option) =
          let updatedDecisions = updateDecisions (model.UserData.ImportDecisions) sheetPath decisionOption
          dispatch <| SetUserData {model.UserData with ImportDecisions = updatedDecisions}
`

Looking at the SetUserData message, it seems that when it is detected, a new model is returned with the updates. I'm certain that I'c doing something wrong, but I've been looking at other examples in the codebase and can't quite point out what exactly it is. I think I'm not quite understanding how to get hold of the new model each time and update that.

Any help would be much appreciated.

Thank you.

tomcl, 2023-08-12:
OK – don’t change UserData – it is meant for persistent state that is loaded and saved with a sheet.

The whole of the Issie model is available. You have two options:

  1.  If the info is needed only while the popup is displayed – add your map as an extra field to Model.PopupDialogData
  2.  If the info is needed to persist without a popup add it as an extra field to Model.

There is an issue which is that popups do not have access to the whole of model state.

They currently get passed only PopupDialogData state, as you can see from the viewPopup function. This was, from Marco, a principled bit of information hiding – popups should operate on their own data – which in reality has caused a lot of trouble.

https://github.com/tomcl/issie/blob/5269fa51b9e899ef6ae67720fedc533e0adf0416/src/Renderer/DrawBlock/PopupHelpers.fs#L726-L737

If you pass stuff to a popup from model it will be the data in model when the popup was created – not the most recent model. Remember, this is FP. To get the latest model you need to call the popup function again with a new model. But if you look in mainView popup functions get called with dialogdata but without model. So you can change the model in a popup – but will not see the new model (except for popupdialogdata).

This is a long-standing gotcha – and I guess what I should do is rewrite the popup stuff passing model to the popup function as an extra parameter. That would be quite easy – and would solve the problem.

If however all you need is to read and write data while the popup is present – and then write out to model via dispatch at end – you can add to the popupdialogdata record. You need to create a message to change your field it in.

Does that make sense? I said it previously but it is a subtle point because the very tangled way in which popups are generated makes it hard to see what is “baked in” and what is passed in answer every time the view function is called.

In fact – given the universal head scratching this causes – I think I really should refactor the popup code this way – or if you feel brave you can do this. It would mean changing the type of PopupViewFunc to  (Msg -> Unit) -> Model -> ReactElement  - the current PopupDialogData input can be found then inside the popup as model.PopupDialogData.

To keep all compatible means going through PopupHelpers refactoring, and then changing every popup creation to be compatible. The compiler guides you and compatibility is just changing popupData to model and adding let popupData = model.PopupDialogData at the start of every such function.

Let me know which way you want to go with this:

  1.  Use PopupDialogData
  2.  Use model – I refactor
  3.  Use model – you refactor

If 2 or 3 give me a little while to investigate the refactoring issue first – just to make sure it works as it should.

Tom

The refactoring should be a separate commit from anything else.

IlanIwumbwe, 2023-08-12:
What you have explained makes sense now. Since I only need the info while popup is displayed, I'll add a new field to popupDialogData. I'll go with option 1.

Thank you very much

IlanIwumbwe, 2023-08-24:
**Update**

I've been working on functionality to import dependencies as well. 

[image]

It seems to work, but has a major problem, which is that it is rather ineffecient. 

- I load all component files from the source directory, which I think has to be done in order to detect all dependencies. If directory is large, popup takes longer to load due to this.
- Maybe what I can do instead is write a function which starts from some sheet, loads all custom components, finds each custom component and loads its custom components, recursively until we're at tree base. this effectively traces specific paths through sheet tree instead of loading full sheetTree then finding path of dependencies, which is what is done using your getSheetTree function.

tomcl, 2023-08-24:
Amazing this is such a complex issue!

You are right. Loading time is just v long, and can’t do much about that. But you can load selectively as you say, which will eliminate the problems.

Thanks, Tom

[image]<https://user-images.githubusercontent.com/56346800/263000416-31962c2a-9011-4bf8-91c4-a2536d338ba1.png>

IlanIwumbwe, 2023-08-24:
I'll work on loading selectively, either before or after tomorrow's meeting. 

Nonetheless, I've added extra functionality such that: 
- User is told if dependency doesn't exist in source or destination directory for whatever reason
[image]
- Only dependencies that aren't already in destination directory are shown for import decision
- Dependencies are not imported by default. User is told this.
- For something like this, where mux2 appears for multiple sheets, user pressing any one of the mux2 buttons causes all to activate
[image]
- The "OK" button is enabled when all decisions about overwrite or rename are made. Dependency import decisions aren't a must, sheets will just not be imported if user makes no decision.

tomcl, 2023-08-24:
Well done!

One other thing that occurs to me.

It is useful sometimes to have copies of hardware in the same directory. In that case every sheet needs to be replicated with a different name.

So my suggestion is:

If the sheet(s) chosen are from same directory:

  1.  throw up a choice popup saying this and asking to cancel or duplicate
  2.  include a text field in the choice popup: this will be prepended to all the sheet names to make the names with which they are added.
  3.  enable the duplicate button when every prepended name is distinct from all the existing names (that will normally be true for any non-null text but there could be clashes)

Perhaps – also – a box for what is appended to the sheet name, allows any combo.

Note that prepended/appended chars need to be valid chars for a sheet name
Note that all custom component instance names need to be converted.

This would be very helpful for me sorting out big designs!

I suggest though that you do a PR without this feature then add it. Good to get features onto master. Let me know if you think the existing PR is in a good state to add to master as it.

Best wishes, Tom

[image]<https://user-images.githubusercontent.com/56346800/263080057-21d2553d-e276-42f2-bf29-1047131bbfe6.png>
[image]<https://user-images.githubusercontent.com/56346800/263079889-b83151da-6f38-41ef-9b72-48004f1d18cf.png>

tomcl, 2023-08-24:
Feedback on popup visual design:

  *   It will be easier to use (for many files) if you use a table to keep buttons aligned, and separate action from rest. (see e.g. search Table.table for examples of many tables in issie). You can add a header “Action”). All of which will make the cluttered screen much clearer. Every file will then have a row in the table with an action (but may or may not have decision buttons). Can find a table design that fits I guess.
  *   Delete redundant info: e.g. no need to say there are no dependencies – keep simple (normal) cases simple

[image]<https://user-images.githubusercontent.com/56346800/263080057-21d2553d-e276-42f2-bf29-1047131bbfe6.png>
[image]<https://user-images.githubusercontent.com/56346800/263079889-b83151da-6f38-41ef-9b72-48004f1d18cf.png>

IlanIwumbwe, 2023-08-25:
I think I'll submit a new PR after refactoring the code to load dependencies selectively to remove that delay, and also use a table to reduce clatter on the popup. 

I will then add the new feature after this PR.

tomcl, 2023-08-25:
Ok, please check that you are always refactored properly on master.

IlanIwumbwe, 2023-08-25:
Yes, I will

IlanIwumbwe, 2023-08-25:
I've created a new function that does selective searching. I tested on a directory with 28 items, importing 2 files. Before: ~ 7 seconds of loading time. Now : ~1 second. The loading time has now moved, it will be higher the more sheets the user tries to import, but I think it's still quite good. I tried to import the whole directory, and that took ~ 3 seconds to load. 

To add the table separate from the rest of the code, after inspection, I think I'd have to do quite a bit of refactoring due to the way my code was structured. However, the feature is at quite a good level now, so if you agree that UI issues can be sorted out later, I can send in a PR now. 

Thanks.

tomcl, 2023-08-25:
ok you can send a PR now which will allow better feature evaluation. I am however sure that a table is the way to go to make the UI clean.

IlanIwumbwe, 2023-08-25:
Okay. I will send in the PR, then work on using a table for the UI.

IlanIwumbwe, 2023-08-25:
**Update**

[image]

All functionality still works. UI looks something like this now. "Import" buttons swtiched out for togglable checkboxes

tomcl, 2023-08-27:
Not quite yet.
Should be:
Sheet Information Decision Action

Information “sheet exists in destination”

Dependencies could get one line per dependency with info “dependency of …”. Making each line clean.

Actions for each file will be import or rename or ignore

Do not add unnecessary info (e.g. dependents). Need to make things simpler I think.

[image]<https://user-images.githubusercontent.com/56346800/263322909-1afc24a5-5e55-4139-aa6c-b1a2ffeec886.png>

IlanIwumbwe, 2023-08-27:
Right, thank you very much

IlanIwumbwe, 2023-08-30:
I've got a new UI made, with dependencies having their own rows. All dependencies appear under sheet that depends on them, for each imported sheet. 
[image]

I also added a dialog popup to duplicate a sheet if the user tries to import from current directory. Instead of renaming each instance of the duplicate in other sheets, wouldn't we want those other sheets to maintain old sheet, and yet at the same time we have this new component we can use with its own name?

If user tries to do it with multiple sheets, what do you think about having a row for each sheet, and waiting for user to give prefix for all sheets?

tomcl, 2023-08-31:
closed by #384

closed 2023-08-31 by tomcl