tomcl/issie #309: "Not connected" component

tomcl, 2023-07-14:
Currently when outputs are not wanted they must be terminated with named viewers to prevent wiring errors.

To deal with this a "not connected" component which displays as a T termination connection would be useful. It has a label which is always empty. It has one input port, and no output ports. It does nothing.

This is fairly simple to implement.

ccrownhill, 2023-07-19:
Could I be assigned to this issue?

ccrownhill, 2023-07-19:
Thanks for the hints. I already got started using the Wiki page but and hence forgot about JSONComponent.ComponentType.
I will add that and then I can update the Wiki page.
However, I think by default Wiki editing is allowed for collaborators only, so you might have to change that setting.

ccrownhill, 2023-07-20:
One question about the label: I tried to remove a label by using `createComponent` with an empty string as the label.
However, this will automatically assign different numbers for different instances of the component.
In the type definition of `Component` it says that all components must have a label but it can be empty.
Could you give me a hint where in the Code I can actually let the label be empty?

tomcl, 2023-07-20:
You can use create\component and then update the resulting record.

Maybe create component updates the symbol map, in which case to update the created component you have to update a symbol in the map. Which is a pain. So you could make the label parameter to createcomponent a string option with the (XML doc) spec that a None => add automatic numbering? Then empty label => really empty.

The compiler will easily make you update all the use cases here that need to be changed..

ccrownhill, 2023-07-20:
I found out that the number is assigned in SymbolUpdate.fs in `generateLabel`.
I just updated it to treat the `NotConnected` component like `IOLabel`:

```fs
let generateLabel (model: Model) (compType: ComponentType) : string =
    let listSymbols = List.map snd (Map.toList model.Symbols) 
    let prefix = getPrefix compType
    match compType with
    | IOLabel | BusSelection _ | NotConnected -> prefix
    | _ -> prefix + (generateLabelNumber listSymbols compType)
```

I thought that this was easier to do than change the parameter type of createcomponent since this `match` already existed.

The only problem I had with this is that when I select the component in Issie it still displays a very tiny green square on top of it (for the empty label) instead of nothing.
So I changed the `addComponentLabel` function in `SymbolView.fs` to not draw a box if the label is just an empty string by adding this `when` statement:

```fs
| false, "lightgreen" when comp.Label <> "" ->
```

tomcl, 2023-07-20:
Looks good! Well done.

ccrownhill, 2023-07-20:
The other problem I have is with the simulation code.
When I connect the `NotConnected` component and then go to the Simulations tab I just see a blank page.

The problem is that I am not 100% sure what the following functions should return for the new component:

Function | Return value I am using
-|-
`Builder.fs`: `getDefaultState`|`NoState`
`FastCreate.fs`: `getPortNumbers`|`1,0` (`ins` and `outs`)
`FastCreate.fs`: `findBigIntState`|`fc.OutputWidth 0 > 32, None` (same as IOLabel, but I'm not completely sure I understand this function)
`FastReduce.fs`: `fastReduce`|same as `IOLabel`
`Verilog.fs`: `getVerilogComponent`|`""`

Moreover, in `FastCreate.fs` I set these variables to `false` but I am not sure what they should be:

```fs
let addDriver, addWave =
            match fc.FType, pType with
            | IOLabel, PortType.Input
            | Input1 _, PortType.Input
            | Viewer _, PortType.Input
            | NotConnected, PortType.Input
            | Output _, PortType.Input -> false, false
```

and from `WaveSimSelect.fs`:

Function | Return value I am using
-|-
`getInputPortName`|`""`
`getInputName`|`failwithf "NotConnected should not occur in getInputName"`
`getOutputName`|`failwithf "NotConnected should not occur in getOutputName"`
`getOutputPortName`|`""`
`selectWaves`|this one is mentioned in the Wiki but I can't see where this should be changed
and `WaveSimHelpers.fs`: `getCompDetails`|`"What? can't happen", true` (`descr` and `oneLine`)

These are all the functions from the Wiki page and since my knowledge of the simulation code as a whole is still limited I am
not sure what exactly is the problem here.
Could you give me a hint whether I am missing something?

tomcl, 2023-07-20:
I will gte to this over weekend (before if I have time).

ccrownhill, 2023-07-20:
Thank you very much.

Note about this little update: I realised that the blank screen output was from a state of the program where I did not implement all functions from above.
After setting fast reduce to return an empty `unit`: `()` I got the following problem: When I press Start Simulation in a step simulation nothing happens.

tomcl, 2023-07-20:
I was going to reply with something you can do in this (or similar) cases.

  1.  Ctrl-shift/i  in Issie opens dev tools
  2.  Look for the first red backtrace from an exception
  3.  At the top of the backtrace is the innermost function call – go down it till you gte to the first F# (.fs) file.
  4.  Click on that to be shown the F# like in dev tools – or just got to that file and linenumber directly in your editor.
  5.  Find the bit of code that is making the exception (very few bits iof F# code can give exceptions)
  6.  Correct it.

This should really be written up properly on the wiki…

tomcl, 2023-07-20:
In addition:

Yes, always to “rebuild renderer” in VS etc to check there are no match warnings => missing cases.

Check for errors as in my previous e-mail

What circuit is not doing step simulation? It is killed by addition of your new component??

ccrownhill, 2023-07-20:
Thanks for the tip. However, there are no exceptions when I try to simulate it.
Just when I press the "Start Simulation" (step simulation) button the nothing happens.

The circuit is using an Nbit adder component with 2 1-bit inputs a constant 0 for CIN and an output Y at the SUM output.
COUT is connected to the NotConnected component.
When I replace it with a wire label I can simulate the circuit.

[image]

However, the result is also wrong in that case since the output stays at 0 at all times.
I tested this with a separate clone of the code and found out that this is not the case with the 4.0.1 release.
I will report this as a separate Issue.

tomcl, 2023-07-20:
One issue – step simulation uses only outputs and viewers as outpiuts, and if you have NO outputs it will maybe do nothing. (In that case there should be an issue for an informative warning). The NC component will noty show up as a step simulation output.

Can’t say more since your image does not display. You need insert -> picture not link -> picture.

Best wishes, Tom

[image]<https://user-images.githubusercontent.com/56307024/254954899-717d0444-2c42-4d20-a3a9-59835305dba0.png>

tomcl, 2023-07-21:
simulation problem closed by #324 
This issue kept open for discussion of more NC component issues.

**Notifications or popups**

(1) NC component attached to a port which can be removed using properties should give a notification saying how to do that. Or, even better, a confirmation popup offering to do it (by deleting NC component and the relevant port).

(2) When simulating a circuit fully connected except with some not connected outputs a confirmation popup should list the unconnected outputs and offer to connect them - if so adding NC popups

(3) In the same line: when simulating a circuit which has not connected inputs which can be removed through properties (e.g. adder Cin) a popup should offer to do this.

1,2,3 are all helpful because the highlight better ways to do things which might otherwise not be recognised by user.

ccrownhill, 2023-07-21:
After pulling the fix for issue #324 the step simulation worked correctly.
(It is strange that you can't see the picture because that's the only way to add pictures in Github comments)
After adding a case to `fastReduceFData` truth tables are now working as well.

Next I will have a look at implementing the Popup functionality you suggested.

tomcl, 2023-07-21:
Re pics.

If you have repo access you can upload a png pic to `./docs/img` and then reference it by URL. Or you can upload to your own web space, or a public online picture repo, and do the same. It is a bit of a pain.

tomcl, 2023-07-21:
@ccrownhill Could you update any inconsistencies/ommissions in the Issie component addition wiki page based on your experience?

tomcl, 2023-07-21:
@ccrownhill And, you could PR the completed component before doing the enhancements - since the component on its own is useful.

ccrownhill, 2023-07-21:
This is the pull request for the addition of the component: #325

I will update the Wiki now and then do the enhancements.

ccrownhill, 2023-07-25:
> simulation problem closed by #324 This issue kept open for discussion of more NC component issues.
>
> **Notifications or popups**
>
> (1) NC component attached to a port which can be removed using [...]

@tomcl I have implemented (1), (3) not with Popups but by adding a button below the message that is shown after you press "See Problems".

However, I have a problem with (2) because it is not that easy to add both a wire and a Not Connected component to a port because I need to know the ID of the component before I can dispatch the message to create the wire which means that I need to dispatch the wire creation message inside the Update function which is not that easy because the types for dispatching a BusWireT message don't match when doing this inside SymbolUpdate.fs.

I was wondering whether I should just list the components in this case or find a way to insert the Not Connected components?

tomcl, 2023-07-25:
When using wires and symbols it is best to write the code in buswire (or even sheet). Then you have access to everything should be easy to do this.

tomcl, 2023-07-31:
Almots done now I think.

NC placement – needs to be on same edge of component as port (portOrientation). Now it is always RHS – but the component can be rotated. The NC component needs also to be rotated based on the edge of the port it is added to.
Waveform simulation – button is green for Start when there are NC errors – I think it should be yellow for “see errors”.

**BUG**: width errors  make both the simulators crash, even if there are no NC errors.

For step simulator this happens on L477 of SimulationView  bolded below
let error =
        **let comp = getComponentById simError.ComponentsAffected[0]**
        match simError.ErrType with

tomcl, 2023-07-31:
I've fixed the bug.

Your PR is now rebased, squashed and on master as a single commit.

tomcl, 2023-08-12:
I have been rationalising the main Issie update function, tidying things up.

I have moved your useful message code `TryStartSimulationAfterErrorFix` as a function `tryStartSimulationErrorFix` to `SimulationView` which is a more rational place for it, and leave the update function uncluttered.

To do that I have had to move some helpers earlier in compiler order: to `ModelHelpers` or `SimulationView`. I have also made some minor changes to `TryStartSimulationAfterErrorFix` code (now in `SimulationView.tryStartSimulationErrorFix`):

* Use optics more
* Use local helper functions `with...`  (defined at top of `tryStartSimulationAfterErrorFix`) to handle the addition of commands to the returned model nicely.
* Changed module prefix for references I have needed to move.

The result is less tangled than it was before - a bit - at some point what I think is needed is proper XML comments on all the `SimulationView` functions which call each other. And maybe moving some more stuff to `SimulationView`. Did you find a lack of this?

In case you have dev work changing `TryStartSimulationAfterErrorFix` the rebase may be quite messy. (Not sure, it may be OK).

A side thought. I looked through the code trying to work out could it be simplified. We have three different simulations all using the same simulator in different ways and you would have thought the error fixup be separated from other variabilities in the code starting each one. But I cannot see a nice way to do this. There are good reasons for the differences.

** My refactor in is on commit 066ad8cdf0b90d0a30672136574f13cfa3003696  (on dev-tc). Could you try rebasing on this, if you have work in progress, and let me know if there are problems, or if all is OK?** When I have heard back I can move this change to master.

tomcl, 2023-08-12:
Sorry - dev-tc got a bit tangled up. The correct fix for you to rebase on is `dev-tc-next`  and the commit with the changes is 63548203e1395edcc0c2c5f6ffe70e3c5d717372

ccrownhill, 2023-08-13:
That seems like a good improvement to me.
I tried rebasing onto `dev-tc-next` but that branch doesn't exist and the commit you provided the link to is on the master branch.
Should I just rebase onto master?

tomcl, 2023-08-13:
Sorry - yes I have now put on on master with some other stuff! Let me know if you get into trouble rebasing - there are solutions.

Try rebasing on a tmp branch, so you can undo it if need be.

Tom

ccrownhill, 2023-08-13:
I tried to just rebase and got a couple of merge conflicts even though I haven't done any further work since I was on vacation until today:

```
Auto-merging src/Renderer/Simulator/CanvasStateAnalyser.fs
CONFLICT (content): Merge conflict in src/Renderer/Simulator/CanvasStateAnalyser.fs
Auto-merging src/Renderer/Simulator/DependencyMerger.fs
CONFLICT (content): Merge conflict in src/Renderer/Simulator/DependencyMerger.fs
Auto-merging src/Renderer/Simulator/Fast/FastReduce.fs
CONFLICT (content): Merge conflict in src/Renderer/Simulator/Fast/FastReduce.fs
Auto-merging src/Renderer/Simulator/Fast/FastRun.fs
Auto-merging src/Renderer/Simulator/SimulationGraphAnalyser.fs
CONFLICT (content): Merge conflict in src/Renderer/Simulator/SimulationGraphAnalyser.fs
Auto-merging src/Renderer/Simulator/Simulator.fs
CONFLICT (content): Merge conflict in src/Renderer/Simulator/Simulator.fs
Auto-merging src/Renderer/Simulator/SimulatorTypes.fs
CONFLICT (content): Merge conflict in src/Renderer/Simulator/SimulatorTypes.fs
Auto-merging src/Renderer/UI/SimulationView.fs
CONFLICT (content): Merge conflict in src/Renderer/UI/SimulationView.fs
Auto-merging src/Renderer/UI/TruthTable/TruthTableView.fs
CONFLICT (content): Merge conflict in src/Renderer/UI/TruthTable/TruthTableView.fs
Auto-merging src/Renderer/UI/WaveSim/WaveSim.fs
CONFLICT (content): Merge conflict in src/Renderer/UI/WaveSim/WaveSim.fs
error: could not apply 279d49fa... add listing of problematic components in error message
hint: Resolve all conflicts manually, mark them as resolved with
hint: "git add/rm <conflicted_files>", then run "git rebase --continue".
hint: You can instead skip this commit: run "git rebase --skip".
hint: To abort and get back to the state before "git rebase", run "git rebase --abort".
Could not apply 279d49fa... add listing of problematic components in error message
```

Because I don't have local work should I just restart further development from a master copy or try to fix these conflicts?

tomcl, 2023-08-13:
Ok. If you have no work not from the PR then yes it is easy - just take a new copy of master and work from that! You can delete your old dev branch and create a new one with same name, or juts use a different name.

tomcl, 2023-08-13:
One of my commits was adjusting your code. Basically, if components were deleted after simulation start while viewing errors the code would crash when looking up non- existing component or connection ids. I did some correction of the obvious cases, there may however be some cases not corrected - though I have not seen that.

ccrownhill, 2023-08-13:
> Almost done now I think.
> 
>     * NC placement – needs to be on same edge of component as port (portOrientation). Now it is always RHS – but the component can be rotated. The NC component needs also to be rotated based on the edge of the port it is added to.
> 
>     * Waveform simulation – button is green for Start when there are NC or other errors – I think it should be yellow and say “see errors”.
> 
> 
> **BUG**: width errors make both the simulators crash, even if there are no NC errors.
> 
> For step simulator this happens on L477 of SimulationView commented below
> 
> ```
> let error =
>         let comp = getComponentById simError.ComponentsAffected[0] // out-of-range EXCEPTION
>         match simError.ErrType with
> ```

Ok, I will do that and then fix the two problems mentioned in the post quoted above.

> One of my commits was adjusting your code. Basically, if components were deleted after simulation start while viewing errors the code would crash when looking up non- existing component or [...]

I will have a look at this, too.

ccrownhill, 2023-08-13:
> One of my commits was adjusting your code. Basically, if components were deleted after simulation start while viewing errors the code would crash when looking up non- existing component or [...]

Could you tell me which commit that was so I can find the changes more easily?

tomcl, 2023-08-13:
https://github.com/tomcl/issie/commit/fde9074c907bdc511b6df4abaeaf2480c8e51e80

FYI - it is obvious from the commit name!

ccrownhill, 2023-08-13:
Yes, I am sorry for not replying again. I had already found it by name

ccrownhill, 2023-08-17:
I have now fixed the last two problems in this pull request: #356 

However, currently there is a merge conflict because `dev-kc` in this repository is probably not up to date with `master`.
What should I do about this?

ccrownhill, 2023-08-17:
Moreover, I am not quite sure about how well I did functional decomposition and if it could be done better in the following case:

I moved the code to create a new wire in to the function `newWire` (in `BusWireUpdate.fs`) so that I can use that when creating a new connected NC component in `SheetUpdate.fs`, as well.
However, because the Cmds from `SheetUpdate.fs` and `BusWireUpdate.fs` I had to also add a second generic function for converting the message given by `newWire` to the correct Cmd type:

```fs
let cmdMapFromMsgOpt
    (mapFn: 'b -> Cmd<'a>)
    (msgOpt: 'b option)
    : Cmd<'a>
    =
    msgOpt
    |> Option.map mapFn
    |> Option.defaultValue Cmd.none

let newWire inputId outputId model =
    let wireId = ConnectionId(JSHelpers.uuid())
    let nWire =
        {
            WId = wireId
            InputPort = inputId
            OutputPort = outputId
            Color = HighLightColor.DarkSlateGrey
            Width = 1
            Segments = []
            StartPos = { X = 0; Y = 0 }
            InitialOrientation = Horizontal
        }
        |> smartAutoroute model

    if Map.exists (fun wid wire -> wire.InputPort=nWire.InputPort && wire.OutputPort = nWire.OutputPort) model.Wires then
            // wire already exists
            model, None
        else       
            let newModel = 
                model
                |> Optic.set (wireOf_ nWire.WId) nWire
                |> BusWireSeparate.updateWireSegmentJumpsAndSeparations [nWire.WId]
            newModel, Some BusWidths
```

I this good practice or can I do this better somehow?

tomcl, 2023-09-03:
Re the above - I think you just need a version of dispatch that will send a different type of Msg (see examples in issie) or some specific converters (again see draw block updates functions) that will accept a given Msg and return the appropriate Cmd. you cannot narrow commands that way (Symbol Msg will not fit into buswire Msg) but buswire code can now send symbol messages directly if it wants.

Abstracting to a generic map (which includes None) is I think juts not helpful - it does not make things more readable? the converters are so simple it is not really worth it.

tomcl, 2023-09-03:
Has this now had its final PR? Can i close?

closed 2023-09-10 by tomcl