tomcl/issie #314: Add a demo design to Issie

tomcl, 2023-07-16:
It would be useful to have a demo design (or even several) built into Issie so that they could be loaded via project menu commands "load CPU demo", "load counter demo", etc.

The work to do this is:

- Work out how to read static files "assets" from the file system: see renderer/testAssets and try it out when called from sheet menu running Issie under `npm run dev`.
- Add a directory containing e demo project to static assets
- Using the functions in FilesIO.fs write a file menu command with a submenu that reads one of the demo projects from static assets, write it to normal file system, then load it as normal. Writing to user file system should use "create project" dialog to work out where to put the demo project - maybe with a fixed project name.
- Add a set of demo projects using a submenu to select which one is loaded.

Get ideas from multiple people including me as to what demo hardware in Issie might be useful for.

IlanIwumbwe, 2023-08-31:
May I please be assigned to this issue?

IlanIwumbwe, 2023-09-06:
when loading projects, should I make it such that the loaded demo becomes the current project, or should I just make it the last used directory, such that the user can import the sheet into their current project?

IlanIwumbwe, 2023-09-06:
**Update**

I've got it such that the menu looks like so:
[image]
 - In the future when new demos get created, they each get their own menu option.
 - When selected, a file selector comes up, asking the user for the directory into which demo should be copied
 - Copying into directories isn't allowed. Directory names already existing isn't an issue, since in file selector, if user types such a name they are redirected to that directory
 - The project is copied over successfully.

The next stage is to make demos and add them to the menu

tomcl, 2023-09-06:
Demos should be loaded, I think, into a special directory inside the Issie executable directory.

tomcl, 2023-09-06:
One thing – there should probably be the facility easily to add multiple demos?

tomcl, 2023-09-06:
How about adding a “open demo project” link to the initial project open menu? And make this go to a chooser screen with (neat – same width) buttons for each demo.

[image]<https://user-images.githubusercontent.com/56346800/266095910-ee7c8c71-4190-468e-99bb-c1644fe91804.png>

IlanIwumbwe, 2023-09-07:
I like the idea of adding an "open demo project" link on the initial project open menu. I will work on this, and add functionality for multiple demos on the chooser screen, which shouldn't be too hard.

tomcl, 2023-09-07:
Yes! The chooser screen can say a bit about each demo, as well…

tomcl, 2023-09-07:
Probably the correct sematics (which needs to be made clear on the popup screen) is:

  *   When you close and reopen a demo project you get the original hardware copied again.
  *   To write your own hardware you should open  anew project.

IlanIwumbwe, 2023-09-07:
I see. At the moment, the user effectively creates a new project but with components from the demo file in some directory they choose. When they load demo project, the hardware will be the initial demo hardware, but if they open the project they created from before, it will contain any changes made like any other project would.

Is this fine, or should it be that saving a copy of the demo project then building onto it if you so wish shouldn't be allowed?

tomcl, 2023-09-07:
The only issue is that to make it v easy to check stuff with demos it might be good to have auto-placed directory inside Issie binaries? Personally it is always a pain for me to decide where to put projects till I have this set up as standard. What do you think?

IlanIwumbwe, 2023-09-07:
Ah I see. I'll set it up to automatically set up a demos directory in Issie executable directory, such that when the user chooses some demo file, it gets copied over from static to this demos directory.

tomcl, 2023-09-07:
Yes. I am not 100% sure about this. But I think so. With the idea that choosing a demo again simply overwrites any previous copy.

IlanIwumbwe, 2023-09-07:
Yes.

I may be missing something, but I can't get popups to work from the initial unclosable popup menu. I keep getting exception errors, but I'm not sure where they are coming from.

tomcl, 2023-09-07:
Ah – yes maybe they are called from the view function only when project is open.

IlanIwumbwe, 2023-09-07:
I thought that as well, so I moved the button under the Project tab, such that at that point, there's an open project, but that didn't work either. I funnelled it down to this function being the cause of the error:        
let demosList =
            div [] [
                Table.table [] [
                        thead [] [ tr [] (List.map headCell ["Demo" ;"Information"; "Decision"]) ]
                        tbody [] (  FilesIO.readdir demos
                                    |> Array.map (fun path -> 
                                        demoContent path ) )
                       
                ]
            ]
But I'm not sure why it doesn't work

tomcl, 2023-09-07:
I will look at this this evening.

IlanIwumbwe, 2023-09-07:
Alright. Thank you very much

IlanIwumbwe, 2023-09-08:
**Update**

Menu looks like this:
[image]

Popup looks like this:
[image]

Loading choice:
[image]

Pressing "Open Demo" loads demo as current project.

tomcl, 2023-09-08:
Ok, so these things are a matter of personal opinion in part.

  1.  Re popup – I would have clickable rows same as initial screen (with cancel – but no OK). Each row could be:
  2.  Project name, description, number of components, number of design sheets

For what demos it would be good to have one demo – it could be the eep1 register file which currently has v complex wireing – rearranged using lots of wire labels. Since people do not know much about them.

Re recents; I can see why you like this but I am on side of not making it easy to download and adapt demos. I feel it is complex. I would juts leave them out of recents so that changes are never persistent. I realise this is opinionated.

[Screenshot (90)]<https://user-images.githubusercontent.com/56346800/266667629-c1c67d83-8689-4a30-af79-7cf88b7268a6.png>

[Screenshot (92)]<https://user-images.githubusercontent.com/56346800/266667707-10042acc-fefd-496d-a57e-22db7ffe37e1.png>

[Screenshot (93)]<https://user-images.githubusercontent.com/56346800/266667783-4efae7b6-99c2-4748-9a8d-26af217692d7.png>

IlanIwumbwe, 2023-09-08:
Alright.
Thank you for the feedback

IlanIwumbwe, 2023-09-09:
**Update**

The demo project select menu now looks like this:
[image]

Clicking the row makes that demo the current project, without adding the directory to the recents listing. 

The register file is the same one from eep1, I just added a lot of wire labels:
[image]

tomcl, 2023-09-09:
Demo project popup – great. I hope you agree with me that simple & clean UIs are good!

Wire labels: surely the real merit of wrie labels comes when you label each of the outputs – and each of the inputs – avoiding all of that wiring?

[image]<https://user-images.githubusercontent.com/56346800/266779963-fe89b851-e88a-45ec-88fe-0fb26baf7088.png>

[image]<https://user-images.githubusercontent.com/56346800/266780156-21450f73-403d-4902-ba2f-caaa459e1766.png>

tomcl, 2023-09-09:
Oh – suggest “register regx16x8 file from eep1 demo using wire labels to simplify wiring”

[image]<https://user-images.githubusercontent.com/56346800/266779963-fe89b851-e88a-45ec-88fe-0fb26baf7088.png>

[image]<https://user-images.githubusercontent.com/56346800/266780156-21450f73-403d-4902-ba2f-caaa459e1766.png>

tomcl, 2023-09-09:
(sorry)

Suggest – fulladder (simpest) fist – then eep1 – then simplified register file?

[image]<https://user-images.githubusercontent.com/56346800/266779963-fe89b851-e88a-45ec-88fe-0fb26baf7088.png>

[image]<https://user-images.githubusercontent.com/56346800/266780156-21450f73-403d-4902-ba2f-caaa459e1766.png>

IlanIwumbwe, 2023-09-09:
Yes, clean UIs are great!

Oh, I completely forgot how to use wire labels to simplify wiring. That's a silly error, I will correct this.

I agree with the demo listing, this will be implemented.

IlanIwumbwe, 2023-09-09:
[image]
New register file design

tomcl, 2023-09-09:
Yes! I think on balance I’d do Din as wires without labels – no crossing is needed with all the other labels.

[image]<https://user-images.githubusercontent.com/56346800/266782194-385df909-c237-4dc8-b013-16aafd4923df.png>

IlanIwumbwe, 2023-09-09:
Oh yes, that makes sense

tomcl, 2023-09-09:
Or – maybe I would keep DIN as wire labels and use wires (no intersections)for the enables?

[image]<https://user-images.githubusercontent.com/56346800/266782194-385df909-c237-4dc8-b013-16aafd4923df.png>

IlanIwumbwe, 2023-09-09:
I like having DIN as wires, since I think it is intuitive to see that the same data goes to the inputs, while the enable bits are decided upon by the input address.

closed 2023-09-10 by tomcl