tomcl/issie #323: Step simulation output not changing

ccrownhill, 2023-07-20:
**Describe the bug**
I found the problem with this circuit which should output 1 in this simulation setting but Y stays 0:

[image]

Then I tested all gates by themselves which didn't work either:

[image]

It didn't work for any other gates either. Note that for NOT, NAND and NOR it was constantly HIGH.

**Issie version**
This problem only happened with the current master branch version (commit: d940113) and not when I used the 4.0.1 version.

**Desktop (please complete the following information):**
 - OS: Linux Mint 21

tomcl, 2023-07-20:
This is a bug introduced from the new simulation. We thought it was fixed - but obviously not. It works now for clocked outputs but not combinational as here. (I tested - even combinational outputs in a circuit with clockjed outputs which is clocked never change from 0.

@0x6770  may be able to see quickly what is wrong and fix it!

0x6770, 2023-07-21:
This error was due to a misuse of `FastRun.runCombinationalLogic` in `FastRun.changeInput` which was introduced in a fix last week.

closed 2023-07-21 by tomcl