tomcl/issie #327: Small easy improvement to waveform simulation UI

tomcl, 2023-07-21:
**Is your feature request related to a problem? Please describe.**

Waveform simulation UI is unclear when sheet simulated is combinational

**Describe the solution you'd like**

1. (easiest solution). After attempting to start the waveform simulation on a combinational sheet the UI should not say "simulation started with a long text beneath.
2. (Better solution). When the waveform simulation tab is open on a combinational sheet the start button is greyed out and the "combinational circuits can only be simulated using step simulator message is displayed (as it is now when an attempt is made to run the waveform sim).

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

tomcl, 2023-09-14:
>   *   Is it a good idea to get rid of that long text that overflows and showing that information on the info popup, which the user can click if they don't know how to use the popup?
>   *   I'm not sure whether I am missing something, but I don't know if the better solution is possible, since the state of the wavesim is only ever updated as 'NonSequential' once the start simulation button is clicked. Looking at the WaveSimModel type, I couldn't see other things I can check to imply this, maybe if I check all sheets or something like that. I think it might be easier to just do away with the long text, tell the user the circuit has no sequential logic, and change the Simulating sheet '[sheetName]' to something like Cannot simulate sheet '[sheetName]'

I need more context to understand this?

tomcl, 2023-09-14:
> Is it a good idea to remove this overflowing text, and add it to the help popup that comes up when the user presses the blue 'i'?
> [image]<https://user-images.githubusercontent.com/56346800/267909985-8fbbbf8f-9005-4235-82ef-b749de97bbd4.png>
>
> And change the top bold text to 'Cannot simulate '[sheetName]''
>
> This is the definition for WaveSim
> [image]<https://user-images.githubusercontent.com/56346800/267910428-c03c5482-9b41-4057-b9ae-80d20497a2ca.png>
>
> My initial thought was to use this to check whether a circuit was combinational, but before you press the 'Start Simulation' button, the state is 'Empty'.
>
> -

Ah, yes.

I think:

  *   Delete "sheet" from title to simplify it
  *   Remove (i)
  *   Delete all text
  *   Add RH context menu item for help to title & text below
  *   If there are no waveforms and clocked logic change the message after starting simulation to "Use Select Waves to add waveforms. Right-click for help."
  *   Remove select RAM and negotiate with Samuel re adding it to the waveform selector.
  *   Swap the simulation & refresh buttons - remove refresh button (don't just inactivate it) if circuit has not changed.
  *   Add right-click help to all of cursor control explaining it.

What do you think?

IlanIwumbwe, 2023-09-14:
That sounds good

IlanIwumbwe, 2023-09-16:
Good evening Dr Clarke. I've been trying to add new context menus for help on the waveform simulator, but I'm not sure why I can't get it to work.
I added my context menu here:
[image]

But it isn't part of the context menus since I get this error: 
[image]

Is there anything I am missing that can help?

Thank you

tomcl, 2023-09-17:
You need to add a menu attachment to UpdateHelpers.getContextMenu  – which examines mouse posn and works out which menu.

type RightClickElement can be added to.

[image]<https://user-images.githubusercontent.com/56346800/268462910-ce8948a8-3068-41fb-9cee-5c7c9e67f08d.png>

[image]<https://user-images.githubusercontent.com/56346800/268462941-6fc7a4e3-f532-432c-a970-66a373889110.png>

IlanIwumbwe, 2023-09-17:
I had already done that. 
I added 'WaveSimHelp' to the DU:
[image]

Then added the attachment :
[image]

Then added it under this match statement

IlanIwumbwe, 2023-09-19:
Good morning, Dr Clarke. I rebuild Issie dev and the context menu shows up now.

What do you think about adding tabs for each feature under 'Waveform simulation' which have all the explanation about how it works, then maybe saying somewhere: 'Right click on element to navigate to help tab.' Then the tabs could be: 'Waveform and RAM selection, Waveform Operations, Cursor, Miscelleneous. 

[image]

tomcl, 2023-09-20:
> The UI looks like this now:
> [image]<https://user-images.githubusercontent.com/56346800/269227686-98b07e92-cb37-4da4-9170-9b2d0a81bae5.png>
>
> Start simulation with no waves does this:
> [image]<https://user-images.githubusercontent.com/56346800/269227899-54a7a27f-b9c1-4d59-9261-76103333e77c.png>
>
> Context menu shows info for each relevant wave sim feature on popup.
>
> When waves are selected, the rest of the buttons are shown like this:
> [image]<https://user-images.githubusercontent.com/56346800/269228326-90810a67-f6bf-45f1-a196-9fd8c9abce27.png>
>
> Reload button shows up if sheet changes like so:
> [image]<https://user-images.githubusercontent.com/56346800/269228593-0361665a-450d-4f8d-90ea-92cd9919e220.png>

Looks good to me, I think. Are you happy with it?

Oh – check what it looks like if simulating a combinational ckt.

IlanIwumbwe, 2023-09-20:
Yes, I'm happy with it. 

On a combinational circuit:
[image]

tomcl, 2023-09-20:
Yes – that is fine.

So what are you doing next?

[image]<https://user-images.githubusercontent.com/56346800/269232168-9d924bf2-9898-4421-91dd-f1d531cf08df.png>

tomcl, 2023-09-20:
> Probably nothing. I'd like to take this last week and a half off.

😊 understood! You have done a lot of work for issie!

IlanIwumbwe, 2023-09-20:
Thank you very much! I've had a lot of fun working on it. I've also learnt a lot. 🙂

tomcl, 2023-09-20:
clodsed with #413

closed 2023-09-20 by tomcl